RavenSystem/esp-homekit-devices #194: I can't see temperature and humidity Sonoff TH with DHT11 and DHT22

pawelosin, 2019-01-18:
I use three Sonoff's TH16 with different sensors and different software. When i connect ds18b20 sensor to Sonoff with RavenCore software i see correct temperature, but when i connect DHT11 or DHT22 sensors I don't see neither temperature neither humidity. Off course i change settings with the Eve App: 1 for DHT11 and 2 for DHT22. I checked it in thermostat mode and in  Switch Basic + TH Sensor mode. In my DHT sensors are connected resistors 4,7 Kohm between power and data. Now i use version 0.5.14, but version 0.5.8 didn't show me these data too.

seritos, 2019-01-18:
Which GPIOs are you using for switch mode and TH connected?

digiservice2001, 2019-01-20:
I'm using GPIO14 for Temp sensor data pin. And again: everything works well till a reset or restart.
Thermostat works the same... 
I don't know what is the solution. :-(

digiservice2001, 2019-01-21:
Does anyone have this problem? or just me?

pawelosin, 2019-01-22:
I am using standard GPIO definied in OTA version 0.5.14. I connected dht11 to Sonoff th through the micro jack plug.