RavenSystem/esp-homekit-devices #196: sonoff basic with RavenCore v0.5.15，pls help！！

lfygif, 2019-01-19:
I don't quite understand it after a couple of hours of tossing and turning.
Sonoff basic has brushed the latest version of Raven firmware directly connected to homekit, why open it automatically every two hours?
After a long time, I couldn't find the reason. I wondered.
Do you have God's help?

GPL71, 2019-01-19:
whic version of sonoff basic?
did you ty erase all, and flash again?

lfygif, 2019-01-20:
> whic version of sonoff basic?
> did you ty erase all, and flash again?
version of sonoff basic？ 
Yes,i do erase all . again an again

lfygif, 2019-01-20:
@GPL71  Yes,i do erase all . again an again!
PLS help ,Thanks

GPL71, 2019-01-20:
which os are you using? Windows or OSX?
I encoured in problem flashing sonoff with windows
After using OSX solved everything
Don't ask me why ... but using old firmware (before 0.5) windows working as a charm.... Now using osx all is good

St0nehead, 2019-01-21:
macOS Mojave. All works like a charm prior to 0.5.10. 
Has some problem with 0.5.11-0.5.15 (no DHCP, no captive portal, etc)