RavenSystem/esp-homekit-devices #197: Cant connect to shelly 1 local wifi

Idontknowme-18, 2019-01-19:
Hi,

First of all, thankyou for your work - I've been using with sonoff devices for some time and think its great.

I've purchased some Shelly 1's to try out, but am having some problems when connecting to them.

I flash them no problem. Then once powered up on 230v I can see the LCM-xxxxxx network, but when i try to connect, it never connects, so i cant get to the setup screen. I've tried loading a browser on captive.apple.com but still no luck.

Thanks for any help you can offer.

RavenSystem, 2019-01-19:
https://github.com/RavenSystem/esp-homekit-devices/wiki/RavenCore#board-type-default-1

Idontknowme-18, 2019-01-19:
Thanks, Sorry i missed this. All sorted now.

closed 2019-01-19 by Idontknowme-18