RavenSystem/esp-homekit-devices #202: Shelly1 stateless switch

gijsstegehuis, 2019-01-20:
First of all thanks for your great work. Your firmware has been working great!

I am wondering whether i can use my shelly1 as a stateless switch (the documentation suggests its not possible right now). I tried device types 3 and 10, just to be sure, but they don't work (accessory stops responding)

Are there any other ways to achieve this?

GPL71, 2019-01-27:
you need set shelly by UART without 230 VAC
set as following
device type 3
external switch 2
log 0
connected by ftdi to 3.3v
and reboot

after it is connected 230 don't change setup for any reason

Shelly 1 workk great

gijsstegehuis, 2019-02-16:
@GPL71 finally have been able to test this.

Basically what you said is working, but **until** i hook it up to 230AC, because then it stops responding.

So when i configured everything on 3.3v, it seems to be all working fine.
Then when i hook it up to 230AC, it stops responding in the home app.
I then connected it back to 3.3v to check the console output, but that doesn't show anything at all.
Looks like it crashed somehow. Is there any way i can recover the log files from it?

It's worth noting that i didn't connect any load to the OUT-A and OUT-B ports on the shelly.