RavenSystem/esp-homekit-devices #206: IMPORTANT NOTICE!! v0.8 new concepts and roadmap

RavenSystem, 2019-01-20:
Please, read here about v0.8:
https://github.com/RavenSystem/esp-homekit-devices/wiki/TODO-List

BETA testers will be needed.

Any suggestion will be welcome.

rechin304, 2019-01-21:
Count me in.

peros550, 2019-01-21:
One idea. How about implementing a web server that would provide a web page to the user so that the user would be able to change all settings there instead of the eve app?

seritos, 2019-01-21:
Willing to participate

gerarddvb, 2019-01-21:
I can help, I have available for testing;
- 2 Sonoff basic
- 2 Wemos D1
- 1 Shelly 1
- a couple of ESP6288 boards

NegativeKontakt, 2019-01-21:
I am happy to help out, have some sonoff, Shelly 1 and Nodemcu. Particularly intrested in the sensor, looking to possibly make something up with a nodemcu mini and PIR

sercitecotf, 2019-01-21:
yo también estoy dispuesto y disponible :-)

misc2000, 2019-01-22:
I can help, with testing and bugfixing (pull requests)
I have available for testing;
    Sonoff basic, S20, S26,  
    Wemos D1 
    and many more ESP8266 boards and 20+ diferent sensors
Let me know how to I can help

St0nehead, 2019-01-22:
I have:
S20
S26
TH16
Pow
Basic
Dual
T1
SC

f1jsh, 2019-01-23:
it's possible for me 
i have 
SONOFF 4CH PRO
SONOFF 4CH
SONOFF DUAL
SONOFF S20
SONOFF TH16
SONOFF POW
Oliver

jubudu, 2019-01-23:
I am interested in the  beta testing program. The following devices are in use
- Sonoff Th, S20 and touch
- Sonoff 4 channel standard and pro
- Shelly 1 amd 2
- NodeMCU
- Wemos D1
- Witty Cloud
- Electrodragon dual relay board
Kind regards Jupp

vmarks, 2019-01-25:
I have some esp8266 boards and two IFAN02 . (I am -really- looking forward to the future when that's supported, willing to beta test for it.)

pandarek, 2019-01-25:
I will be happy to help you too :)

RavenSystem, 2019-01-25:
I see many people interested in this. Amazing!!! Finally, it will be a new firmware separated from RavenCore.

fcrespo82, 2019-01-26:
I have a Sonoff Basic and a NodeMCU with esp12e 
Willing to help in anyway I can

kittmark, 2019-01-27:
> I see many people interested in this. Amazing!!! Finally, it will be a new firmware separated from RavenCore.

Hello, 2 things I would suggest for this brand new fw
- a mode where the possibility of accidentally sending a sonoff/shelly in maintenance mode by long pressing an external phisical button is reduced by a time limit (like instead of long press 10 seconds for maintenance mode, make it “turn on/off 5 times within the first 60 seconds from power up”). This way people would be more confident to install sonoff/shelly devices around the house. 

- for shelly2, a mode where an external physical momentary button could make the device rotate between 4 modes (on+off, off+on, on+on, off+off), in order to mimic a rotating relay. A lot of people would benefit from this, because this way you know the state. (compared to keeping the legacy rotating relay and creating an homekit momentary switch to excite it). 

- easily create custom named homekit alarms, triggered, among other things, when the dry contacts of a Shelly1 are closed. This way people can go crazy with custom sensor triggered alarms of any kind.

Thanks and keep up the AMAZING work.

greggsymington, 2019-01-27:
I'd love to test the beta firmware. I've got a few SV's and a TH16.

I'd love to see a feature to turn a sonoff into a contact switch using the GPIO for a door or similar.

drpodd, 2019-01-27:
Sign me up for testing! New user with a couple sonoff basic.

fcrespo82, 2019-01-28:
@RavenSystem Do you have any estimated date as for when you will be able to start it?

RavenSystem, 2019-01-28:
@fcrespo82 No, sorry.

dbambus, 2019-01-30:
I got some NodeMCUs and a Sonoff S20 for Betatesting.

pawelosin, 2019-01-31:
Sign me up for testing! I have many Sonoff devices for example: Sonoff basic, Sonoff rf, Sonoff TH, Sonoff T1 (1ch, 2 ch and 3ch), Sonoff dual, Sonoff POW an other: Shelly1, Shelly2, ESP01, Wemos D1 mini.

Partizan7676, 2019-02-01:
if needed can help testing with iFan 02.

fst2112, 2019-02-01:
also very interested in testing !

kalety, 2019-02-13:
Jose, yo quieroooo!!!. Jose I want to be betatester, many devices to test!. Administrator about 800 people telegram domotic group, many of then Ravencore's users...

i3laze, 2019-02-16:
Two more feature requests:

1. Tiny persistent logging, viewable from web-mode.
2. FW V0.6.next for RavenCore with an option to upgrade to new Architect via OTA (don’t wanna wire up all my RavenCore accessories).

RavenSystem, 2019-02-25:
Update status:

- Multiple rooms is done: First accessory works like a bridge too, and rest of accessories are linked to the first accessory. In this way, you can configure each accessory in a different room.
- Actually, I am able to declare an array of 1..N HomeKit switches.
- Configuration is saved in JSON format, to allow dynamic setups.

i3laze, 2019-02-26:
Accessory grouping is a native feature in Home.app. 
What’s the idea then? To hide other accessories and present them as single device?

kalety, 2019-02-26:
> Update status:
>
> * Multiple rooms is done: First accessory works like a bridge too, and rest of accessories are linked to the first accessory. In this way, you can configure each accessory in a [...]

José, how I could test this firmware?

rgnovak, 2019-03-04:
Still looking for beta testers?
Devices that I work with:
Sonoff basic
Sonoff S20
Shelly 1
NodeMCU
D1 mini
gosund SP111

pilot1981, 2019-03-04:
@RavenSystem you can add here new feature to control robot roomba?

apacherg, 2019-03-04:
Great .... When you want

seritos, 2019-03-06:
Lets support @RavenSystem with some HW for his new testings! 

https://github.com/RavenSystem/esp-homekit-devices/issues/272

Thanks!

arjunaukta, 2019-03-27:
Count me in.

sorriso93, 2019-03-31:
count also me (wemos d1 mini, sensors and relays)

caligari74, 2019-04-09:
Still in need of beta testing? Please add me!
Lots of Sonoff and ESP8266 devices here...

pixidixi88, 2019-04-12:
You can count me for beta tests. I have 4 Sonoff TH one basic and one dual. I consider to buy more.

Korcsog, 2019-04-26:
@RavenSystem Is the new HomeKit Accessory Architect still available for beta testing? I've got a lot of custom made esp-12 based boards (garage door opener for example) and want to test RavenCore FW because of that gpio setup method!

tohmc, 2019-05-04:
Hi,

I have
18 shelly 1
4 shelly 2
2 shelly 2.5
8 sonoff Basic
4 sonoff 4ch
And
A lot of wemos d1 with relayshields and Buttons.

Al work Fine with your Firmware.
I will Be a Beta Tester for some devices!

RavenSystem, 2019-05-09:
https://twitter.com/RavenSystem/status/1126535435960950786

RavenSystem, 2019-05-09:
A single esp8266 chip can be very powerful with HAA:
https://twitter.com/RavenSystem/status/1126581263060602883

RavenSystem, 2019-05-13:
Future is now.

peros550, 2019-05-13:
Well done! Many thanks for another great release!

mafyata, 2019-05-13:
Hi I flashed my sonoff basic to test haa. Can you make setup mode to be like in eve app? It is much more simple to setup. 
Json metod is more difficult. At this moment I prefer to stay at ravencore.

peros550, 2019-05-13:
json method provides much more customization than what Eve app offers. 

It would be great if the device could expose a temporarily a web server before HomeKit server kicks in. In this way, all customization could happen easier than json method.

mafyata, 2019-05-13:
I don't mean to use eve app. I mean setup page to be easy like in mqtt.

RavenSystem, 2019-05-13:
@peros550 @mafyata 

No. JSON method allow me to use more power for useful functions. I don't want a beautiful GUI running in the device to configure it, eating CPU, RAM and flash memory. In a future, you will can use a wonderful JSON generator GUI outside your device (Standalone App or web app...), where you will can setup a device without knowledge of JSON; then, you only will must to copy and paste the generated JSON string.

mafyata, 2019-05-13:
It sounds very good.

RavenSystem, 2019-05-13:
I have added more comprehensive JSON examples: https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Templates

vmarks, 2019-05-13:
"can setup a device without knowledge of JSON"

That's exactly my level of knowledge of JSON! excellent!

closed 2019-05-26 by RavenSystem