RavenSystem/esp-homekit-devices #208: what is the lowest temperature homekit

stepanyurkiv, 2019-01-23:
what is the lowest temperature that can be measured in the homekit. I want to measure the temperature on the street.

ymyshko, 2019-01-23:
ds18b20
When temperature less 0C (negative value) Home app and EVE show 200C

RavenSystem, 2019-02-28:
It seems that esp-open-rtos driver has a bug. I can not do nothing, I am not the developer of the driver.

ymyshko, 2019-02-28:
> It seems that esp-open-rtos driver has a bug. I can not do nothing, I am not the developer of the driver.

Me too.
And winter ends :)

ymyshko, 2019-02-28:
In esp-open-rtos. 
https://github.com/SuperHouse/esp-open-rtos/blob/master/extras/ds18b20/ds18b20.c
=====================================================
       uint8_t temp_msb = get[1]; // Sign byte + lsbit
        uint8_t temp_lsb = get[0]; // Temp data plus lsb
        uint16_t temp = temp_msb << 8 | temp_lsb;

        float temperature;

        temperature = (temp * 625.0)/10000;
        //debug("Got a DS18B20 Reading: %d.%02d\n", (int)temperature, (int)(temperature - (int)temperature) * 100);
        result[sensor_id].id = sensor_id;
        result[sensor_id].value = temperature;
===================================================

In Sonoff-Tasmota. Tasmota show corretly negative temperature
https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/xsns_05_ds18x20.ino
===================================================
        temp12 = (data[1] << 8) + data[0];
        if (temp12 > 2047) {
          temp12 = (~temp12) +1;
          sign = -1;
        }
        ds18x20_sensor[index].temperature = ConvertTemp(sign * temp12 * 0.0625);  // Divide by 16
 ===================================================
I think, negative temperature > 2047
I'll try recompile and test.

RavenSystem, 2019-02-28:
@ymyshko you are right.

RavenSystem, 2019-03-10:
Issue opened at esp-open-rtos repo: 
https://github.com/SuperHouse/esp-open-rtos/issues/700

closed 2019-03-10 by RavenSystem

RavenSystem, 2019-04-10:
Version 0.8.1 has changed the way to read temp from ds18b20 to be compatible with negative temps because there is a bug in eep-open-rtos driver when using read single function.