RavenSystem/esp-homekit-devices #212: Thermostat

Gio1978, 2019-01-26:
Does thermostat has this  future?  if temp <= 21 it turns on heat... if temp >= 21.2 it turns off heating

digiservice2001, 2019-01-26:
I think +/- 0.5 C hysterisis is better. Most of thermostats work like this. Less on/off switching, and that's comfortable.

  ----- Original Message ----- 
  From: Gio1978 
  To: RavenSystem/esp-homekit-devices 
  Cc: Subscribed 
  Sent: Saturday, January 26, 2019 12:14 PM
  Subject: [RavenSystem/esp-homekit-devices] Thermostat (#212)

  Does thermostat has this future? if temp <= 21 it turns on heat... if temp >= 21.2 it turns off heating

  —

---
Ezt az e-mailt az Avast víruskereső szoftver átvizsgálta.
https://www.avast.com/antivirus

RavenSystem, 2019-01-27:
To avoid switching abuse, increase poll period.

closed 2019-01-27 by RavenSystem