RavenSystem/esp-homekit-devices #221: Power consumption in RC

seritos, 2019-01-30:
Hi Jose,

How much is the power cosumption in RC? Is there any sleep mode or lower power mode? 
For those who have dozens of them intalled ( including myself) is quite interesting.

Thanx

seritos, 2019-02-03:
@RavenSystem, any coments? :)

Thanx!

RavenSystem, 2019-02-03:
No sleep mode and no low power mode because wifi must be active for HomeKit. None of my certified accessories (Koogeek) uses low power mode wifi.

RC is coded focusing in a low resource consumption, but there are mechanisms that uses many resources, for example, software filtering to avoid AC interferences (it is used in external toggle and garage door).

If you want to minimize consumption, remember to disable "show setup" and set log output to 0.

closed 2019-02-25 by RavenSystem