RavenSystem/esp-homekit-devices #225: Homekit connect

cnmarch6, 2019-02-01:
OMG where did it go wrong. 

I flash successfully, saw the LCM-37AA63, click on it and it never connected, no webpage pop up. 

So I flash it again without OTA, this time opened home app and scan the code, waited for 5 minutes and still discovering nothing show up, go to wifi and did see the RavenCore-37AA63

HELPPPPPPPPPPPPP

GPL71, 2019-02-01:
@cnmarch6
Try these steps ...
1 do procedure using MAC, second choice WINDOWS
2 Avoid if it is possible iphone and ipad (obviously only for joing to the net)
3 Click on LCM- and while it is trying connection, unplug usb and after 1 sevond reconnect USB. Usually after a while the web page will open.

Partizan7676, 2019-02-01:
same issue. erasing flash with esptool.py will solve it.

cnmarch6, 2019-02-02:
> @cnmarch6
> Try these steps ...
> 1 do procedure using MAC, second choice WINDOWS
> 2 Avoid if it is possible iphone and ipad (obviously only for joing to the net)
> 3 Click on LCM- and while it is trying connection, unplug usb and after 1 sevond reconnect USB. Usually after a while the web page will open.

working now, thanks.