RavenSystem/esp-homekit-devices #228: Shelly 2.5

GPL71, 2019-02-02:


GPL71, 2019-02-02:
@RavenSystem 
Hello ... Is ravencore compatible with Shelly 2.5?

RavenSystem, 2019-02-03:
I don't know, because I don't know its GPIO map, but I think is the same as Shelly2. Shelly 2.5 has an external UART conector, and I think shelly.cloud will publish GPIO map soon.

i3laze, 2019-02-14:
Got Shelly 2 recently and I already know why it's been discontinued - this needle magic sucks:
[image]

RavenSystem, 2019-02-28:
I will support Shelly 2.5 when available.

closed 2019-02-28 by RavenSystem

greensouth, 2019-04-15:
Yummyyyyy!!

Any news on the Shelly 2.5 support? 

Btw, I had my first experience with Ravencore flashing a Magic Home LED wifi controller and … "$)( I'm really excited and impressed with the results!! Keep on the hard work, it's really impressive.

Now waiting for the 2.5, very excited :)

RavenSystem, 2019-04-15:
@greensouth I am glad you like it!

Shelly 2.5 support was added in version 0.8.3. Remember that you can see changelogs under Releases tab: https://github.com/RavenSystem/esp-homekit-devices/releases

I haven't any Shelly 2.5 yet, so I don't know if there are bugs about that.

greensouth, 2019-04-15:
> @greensouth I am glad you like it!
> 
> Shelly 2.5 support was added in version 0.8.3. Remember that you can see changelogs under Releases tab: https://github.com/RavenSystem/esp-homekit-devices/releases
> 
> I haven't any Shelly 2.5 yet, so I don't know if there are bugs about that.

YOU - MADE - MY - DAY!! Thank you!! I'll be trying it in the next days!

Coño, que bien me lo voy a pasar este puente! :)

RavenSystem, 2019-04-15:
Jajaja, pues dale caña! Y esperemos que no haya fallos. Si los hubiera, abre una nueva incidencia (issue) con los detalles.

greensouth, 2019-04-15:
Ya te digo! este puente intentare motorizar una persiana, con que no me salga ardiendo la casa, me apaño jajajjaja! Soy bastante novato con el tema relés  y tal.

mafyata, 2019-04-15:
> Yummyyyyy!!
> 
> Any news on the Shelly 2.5 support?
> 
> Btw, I had my first experience with Ravencore flashing a Magic Home LED wifi controller and … "$)( I'm really excited and impressed with the results!! Keep on the hard work, it's really impressive.
> 
> Now waiting for the 2.5, very excited :)

Can you give me your contact like facebook or viber to ask you about magic home flashing with ravencore ?

greensouth, 2019-04-16:
> Can you give me your contact like facebook or viber to ask you about magic home flashing with ravencore ?

What do you want to know?

mafyata, 2019-04-16:
Eve setup options and did you get ir remote working with raven core

greensouth, 2019-04-16:
> Eve setup options and did you get ir remote working with raven core

I don't have an IR remote, so I never had to configure it. 

In eve I just set:

```
board = 1
device = 14
color boost = 2%
```

(mine is a RGBW controller so had to add the W gpio also -even if it's unused, if not, you will getting unexpected colors):

```
Pin B = 12
Pin G = 5
Pin R = 14
Pin W = 13
```

For GPIO equivalences, I used this pic [https://user-images.githubusercontent.com/29731130/31029735-2f99db9c-a553-11e7-896c-2f71c3d04551.jpg
](url) at [https://github.com/arendst/Sonoff-Tasmota/wiki/MagicHome-LED-strip-controller](url)

mafyata, 2019-04-16:
Mine is RGB.
Pin B =13
Pin G =12
Pin R =5
Pin W =0
How check my GPIO equivalences ? Are the same like rgbw , but without W ?
[image]

i3laze, 2019-04-16:
Please don’t highjack this dead Shelly 2.5 thread, find or make a new RGBW one.