RavenSystem/esp-homekit-devices #229: "Close Garage door" in Siri command doesn't energegize relay

GPL71, 2019-02-03:


RavenSystem, 2019-02-03:
Maybe it is a bug. Screenshot of your EVE settings, please. I am going to test by myself.

RavenSystem, 2019-02-04:
I don’t understand. Screenshot goes away...

RavenSystem, 2019-02-06:
@GPL71 Not enough free time to test yet.

RavenSystem, 2019-02-25:
I don't see sensor close NC in your setup, so I have tested this bug with "sensor close NC = 0" and all of your settings. I can not reproduce the issue. In my case, Siri always works fine. Maybe the issue is caused by a wrong value of  "sensor close NC". Try setting it to 1 if you use 0, or set it to 0 if you actually use 1.

GPL71, 2019-02-25:
@RavenSystem 
Has opened sensor --->not is present in EVE
Sensor close NC ---> not is present in EVE

Can you tell me how to resolve or can you post the screenshot of your eve?

Thank you

RavenSystem, 2019-02-25:
@GPL71 "Has opened sensor" and "Sensor open NC" are both replaced with "Sensor open". But" Sensor close NC" must be present.

[image]

RavenSystem, 2019-02-25:
I have updated" Sensor close NC" info in Wiki just now (It was outdated):
https://github.com/RavenSystem/esp-homekit-devices/wiki/RavenCore#sensor-close-nc-default-0

dvelardi, 2019-02-26:
You can explain better the issue?

I don’t home if it’s the same issue of my case: when I send open command, Apple home app show garage door always in opening state and if I ask Siri which it’s the state of garage door, Siri say don’t know...

DV
3389305149<tel:3389305149>

link - PRIVACY POLICY. http://www.meadinformatica.it/privacy/

https://github.com/RavenSystem/esp-homekit-devices/wiki/RavenCore#sensor-close-nc-default-0<https://hes32-ctp.trendmicro.com:443/wis/clicktime/v1/query?url=https%3a%2f%2fgithub.com%2fRavenSystem%2fesp%2dhomekit%2ddevices%2fwiki%2fRavenCore%23sensor%2dclose%2dnc%2ddefault%2d0&umid=9a17b87a-e5fc-4484-a54c-f678a8c1ab54&auth=48278bb8638c59b19e6a5c9ee0648ca7e9c732c1-38f23230befdbc9d1f344a04ad4c4de76b8f1bd7>

GPL71, 2019-02-27:
@RavenSystem I did your setup and changed the sensor to 1, but problem not was solved.
I have same issues on 4 different boards.
Any suggestion?

GPL71, 2019-02-27:
@RavenSystem this is configuration on all of 4 boards and I confirm the bugs again ... what's difference from your setup? Thank you
 
[image]
[image]
[image]
[image]

RavenSystem, 2019-02-28:
Bug found. Incompatible types int8 and bool.

pilot1981, 2019-02-28:
Incompatible only with SONOFF SV or also with SONOFF BASIC?

DV

RavenSystem, 2019-02-28:
Fixed in 0.6.2.

closed 2019-02-28 by RavenSystem