RavenSystem/esp-homekit-devices #236: Garage Door not reading sensors on startup

MrCmonster, 2019-02-08:
Great work on this!
I'm setting up the Garage Door opener on a Sonoff SV. I believe I found a bug in the Garage Door function.

Prerequisite:
The device has both an open and a close sensor correctly configured.

Steps to reproduce:
1. Open the door from HomeKit
2. Once the door is fully open, simulate a power outage by unplugging the Sonoff and plugging it back in
3. Go into the Eve app and refresh the door status

Expected result:
The Sonoff should read the door sensor on startup and see that the door is open.

Actual result:
The Sonoff sends an alert that the door was closed even though it is fully open.

I think the hardware_init() needs to read the current state but I'm not sure how to fix it.

Thanks

RavenSystem, 2019-02-11:
Yes, sensors are not read on startup. It is necessary to implement it in future updates.

pilot1981, 2019-03-01:
@RavenSystem  when this feature will be developed?

RavenSystem, 2019-03-10:
No yet.

pilot1981, 2019-03-10:
Different question: why I can’t group two garage door together?

i3laze, 2019-03-28:
You might add a fake switch with a relevant name and hook both doors to it in Automation.
This way I worked around the requirement to unlock an iPhone.

P.S. the only fake switch I know is running HomeBridge alongside. May be new Architect f/w will have one.

pilot1981, 2019-03-28:
I solved using default apple home kit command: "hey siri, open all doors in garage".

@RavenSystem you can close

drakorqs, 2019-05-02:
@RavenSystem thank you for your work!
The obstacle sensor also does not keep the last state.
When the obstacle sensor is on, the garage door cannot be opened from the app. (after power off, or reboot)
https://youtu.be/d51Lr62gfmU

RavenSystem, 2019-05-29:
Done in RavenCore v0.9.0.

closed 2019-05-29 by RavenSystem

jat80, 2019-06-17:
> Done in RavenCore v0.9.0.

Does this mean If the Sonoff has a power outage or reboots it won't trigger the garage door?  I had a power outage when the door was closed and when it came back on it opened the garage door.  Same happens if I cut power to it manually and turn it on again, it triggers the switch to open the garage door.