RavenSystem/esp-homekit-devices #238: I this thing compatible?

Moskvitch, 2019-02-08:
https://www.itead.cc/smart-home/inching-self-locking-wifi-wireless-switch.html

Partizan7676, 2019-02-11:
check TODO list. it will added later.

lizzus, 2019-02-27:
Yes, this is fully compatible! You can check here: https://www.twibuk.com/post/1090_sonoff-bassa-tensione-micro-usb-5v-oppure-ac-dc-7-32v-con-firmware-ravencore-fun.html
Racencore firmware is the best!

pilot1981, 2019-02-27:
@lizzus which sonoff type you used? It could be the best choice for my garage door opener!?

DV

lizzus, 2019-02-27:
@pilot1981 i used this https://www.itead.cc/smart-home/inching-self-locking-wifi-wireless-switch.html
I think this is not the best choice...

pilot1981, 2019-02-27:
I don’t know...using sonoff basic with device type 8 I have the only option to have switch mode, but I need only button mode to send open command to my garage door (garage door close itself) so sonoff 1 channel inching could solve this problem...

DV

GPL71, 2019-03-02:
Yes I confirm, fully compatible, and the inching function is preserved even using @RavenSystem firmware.
You'll need soldering wires for TX RX and GPIO0 to allow flashing

GPL71, 2019-03-02:
@pilot1981 the best board for garage door is Sonoff SV.
I'd used for garage door, and yes it's perfect

RavenSystem, 2019-03-02:
Solved.

closed 2019-03-02 by RavenSystem

pilot1981, 2019-03-02:
@GPL71 I agree with you but open sensor isn’t mandatory, obstructed sensor don’t work yet, if you want integrate power supply and you want a cheaper solution SONOFF BASIC remain the best choice...