RavenSystem/esp-homekit-devices #239: Power monitoring

aslater3, 2019-02-09:
Is it possible to add power monitoring functionality? Either via Eve or native HomeKit?

oremic, 2019-02-09:
chip for shelly 2  power meter MCP39F501.

RavenSystem, 2019-02-28:
Apple HomeKit doesn't support it, and I don't add other characteristics HomeKit.

closed 2019-02-28 by RavenSystem

oremic, 2019-02-28:
Apple homekit does not even support customizations and yet you use EVE to do it.

RavenSystem, 2019-02-28:
I only use custom characteristic to setup accessory. Of course, since this project is open source, you can implement by yourself all options and characteristics that you want.

oremic, 2019-02-28:
Sorry if I was wrong to ask.
The work you did is very great, six, exceptional.
The function of the power meter would be exceptional, unfortunately I can not write a library.