RavenSystem/esp-homekit-devices #241: TH Init state problem

Partizan7676, 2019-02-11:
Try to set it to last state. Thermostat was in heating mode and after power disconnected it comes back with cooling mode. May be last state goes with auto mode? in my case i have current temperature 15C, i set heat to 24C and when i choose auto mode it change it to cooling. My hardware is Sonoff TH 10A with TH sensor

Partizan7676, 2019-02-11:
Test it now - switch to auto mode after few seconds thermostat is switched off

RavenSystem, 2019-02-11:
Auto mode doesn't work and it change to off.

I think that your issue is related to a wrong information. Maybe you are using same options like switches for init state? Check this for init state on thermostat: https://github.com/RavenSystem/esp-homekit-devices/wiki/RavenCore#init-state-th-default-0

And remember that last states are saved after 5 seconds last change, so wait a few seconds until power off your device.

closed 2019-02-28 by RavenSystem