RavenSystem/esp-homekit-devices #242: Wifi Reset Shelly 1

rgnovak, 2019-02-11:
Hi,
the note "Shelly board layout: You must configure board and device type BEFORE connecting Shelly to 230VAC." makes me wonder, how to do a reset of Wifi settings in Shelly 1 if this is already installed (connected to AC power). What does it do if I follow these instructions: Toggle external switch quickly, 10 times in 3 seconds within 2 first minutes since device boots. As far as I understand, this would do a complete reset, so the board and device type would not be set before connecting. Would this destroy the Shelly?
Thanks,
Ralph

GPL71, 2019-02-12:
use EVE app to perform a wifi reset 
To perform it is allowed leave the shelly connected to 230 VAC

RavenSystem, 2019-02-12:
Board type is not changed when a factory defaults reset is performed. Since board type can be changed only manually from EVE App, your Shelly is always safe.

rgnovak, 2019-02-12:
Great, thanks.

closed 2019-02-12 by rgnovak