RavenSystem/esp-homekit-devices #244: Reset and lose possible reconnect Shelly1 v3

koleox90, 2019-02-12:
Hi, 

I have problem with Shelly1 V3.

I have large house and when moving Shelly to another part of house i want change Wi-Fi router. When i set board type on 3 for AC230, and i make a reset Wi-Fi on EVE, lose everything. I must flash firmware again.

How to fix this issue?

GPL71, 2019-02-12:
It's very strange 
I have a large house too and 12 shelly.
I moved them and they work great even when I'd reset wifi

how do you set eve?

koleox90, 2019-02-12:
Maby i do something wrong? 

That my eve setings. 
[image]
[image]

GPL71, 2019-02-14:
set external toggle 2
log 0

do it by ftdi 3.3v ... after connect to 230vac

koleox90, 2019-02-14:
Thanks, I try this settings.

closed 2019-02-28 by RavenSystem