RavenSystem/esp-homekit-devices #245: Button 1 of  SONOFF dual

kalety, 2019-02-14:
Hi!, I'm "Ravencorized" a Sonoff dual R". I use Board type 4, device type 2, external toogles 1 and 2 to type 2. Connect to RX/GND the switch 1 and button1/gnd the second. The first works flawlessly (RX/gnd), the second (button1/gnd) only responses the first time is it's off, the switch turn on, but don't turn off. I'm test the button1 with a button (type 1), and nothing. The pins are ok, because I test it with sonoff tasmota and works well.

Thank you

RavenSystem, 2019-02-14:
Sorry, it is a bug that I found in the code just now. Thanks for your feedback. Next version will fix it, and soon.

kalety, 2019-02-14:
Thank you very much! (Haa?, ;)))

RavenSystem, 2019-02-28:
Fixed in 0.6.2.

closed 2019-02-28 by RavenSystem