RavenSystem/esp-homekit-devices #246: Shelly1 v0.6.1 settings lost

i3laze, 2019-02-15:
I've upgraded my light switch with Shelly v0.6.1. It's been working for 2 days just fine with external toggle: 2.

Just came back home - HomeKit accessory irresponsive, Shelly1 broadcasts it's SSID and is back in Pairing mode.

Is it safe to set it up, while it's hooked to 220AC?
UPD: After re-pairing Eve settings are Board type: 3, but External Toggle1: 0.  Luckily, it was safe.

How to troubleshoot that next time? Could it be some pullup fluctuation, which led to "10 fast switching" reset?

Maybe, it's time to implement some persistent debug logging mode (tiny and with limited duration) sending data to some IoT Cloud?

RavenSystem, 2019-02-28:
> Does current code detects fake double switching (like <100ms)?

Yes.

> How about adding “enable pull-up” option in settings?

Enable internal pull-up resistor is not possible in Shelly devices.

i3laze, 2019-02-28:
Unless it has (at least Shelly 2 does according to the link above).
Also bunch of tweaks mention it here:
https://github.com/arendst/Sonoff-Tasmota/issues?utf8=%E2%9C%93&q=Shelly+pull

Nevertheless I titghened some lousy wires and ghost switching so far doesn’t occur.

RavenSystem, 2019-02-28:
Enabling internal pull-up resistors can burn the esp8266 of your Shelly when connected to 230VAC.

i3laze, 2019-04-30:
I’m still occasionally struggling from shelly1 and shelly2 getting ghost switched.
Yesterday I even got one bulb down.

Any ideas how to troubleshoot this?

i3laze, 2019-05-01:
Looks like I've found the cause of my ghost-switch problem.
These fancy switches do bypass some current to make a LED glow at night:
[image]
[image]

I've swapped them to basic ones - so far so good:
[image]

closed 2019-05-01 by i3laze