RavenSystem/esp-homekit-devices #252: Shelly1 disconnect

gioz73, 2019-02-20:
Hi, I have 4 shelly1 with ravencore 0.6.1 work well but every 3/4days disconnect from network (all together). If I turn off and on the general power work well for other 3/4 days.
Any idea?
Thank

i3laze, 2019-02-21:
Same here.
Today it’s Shelly2 unresponsive. Troubleshooting steps?

RavenSystem, 2019-02-28:
Logs, please.

Fussverkehr, 2019-03-03:
I have the same issues with the shelly 1 and the fritzbox wlan. how can i generat the log files?

RavenSystem, 2019-03-03:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Build-instructions#capture-logs

But for sure it is a wifi network related issue.

peros550, 2019-03-03:
Do you have Apple TV used as hub? If yes, try rebooting it (keeping the router and Shelly devices on) and advice if the issue cleared. I had a similar situation last week where all my Ravencore devices were in no response status. Rebooting Apple TV brought them back.

Fussverkehr, 2019-03-04:
The restart of the appleTV help in my case.

gioz73, 2019-03-05:
Reboot of AppkeTV don’t work for me

closed 2019-04-13 by RavenSystem