RavenSystem/esp-homekit-devices #254: problem with sonoff basic R2 cutted tracks in garage opener scenario

pilot1981, 2019-02-20:
Hi,
I'm trying to use a sonoff basic R2 cutted tracks in garage opener scenario with raven core 0.6.1 but with device type 1 don't saw timer and other useful setting in EVE app (as written in wiki). Then I changed setting to device type 10 but I lost device and reset don't helped me to take again control of sonoff.

So I have two questions:
- how I can take again control of sonoff? erase flash and program it again?
- which is the right device type to use in my scenario?

regards

pilot1981, 2019-02-21:
there is a matrix where find association between device type number and commercial product (sonoff, shelly, ...)?

GPL71, 2019-02-21:
is device 7 garage door

GPL71, 2019-02-21:
check the wiki

pilot1981, 2019-02-21:
it isn't device type 8? also for sonoff basic R2 with cutted tracks?

dvelardi, 2019-02-21:
Solved with device type number 8.

If I have garage door close itself using internal timer, with closed sensor on GPIO2 and without opened sensor: which is right configuration in EVE?

DV

pilot1981, 2019-02-22:
@GPL71 wiki isn't so clarifying for me... If I have garage door close itself using internal timer, with closed sensor on GPIO2 and without opened sensor I need to reset button automatically, I need to obtain this result:
1. initial state: closed
2. manually push button/switch: state change to open
3. automatically reset button/switch and state

I don't need of a switch ON/OFF: I need only a button to push to open a garage...I don't understand howto to come back automatically the switch/button on home app to initial state (closed)....

you know if open sensor feature it's already developed? work fine? in this case I could decide to use a SONOFF SV with more GPIO...

GPL71, 2019-02-22:
I'd use sonoff SV ...
however by EVE you need to set the follow 2 timers

1 duration of push button in HOME APP to lunch the opening
2 working time to open completly the door

Garge door (device 8) has a little bug.
1 is about animation icon
the second about voice control by siri. you need say hey siri open garage door even if you want close the gate

but surely @RavenSystem will correct it

pilot1981, 2019-02-22:
my garage door close itself, I don't need close command... howto you configure other options of stop step and close sensor, for me wiki isn't so clear...

@RavenSystem I ask to you:

- create more clear wiki for duration timer, close and open contact
- how I can configure device 8 in my case? I don't need open command...my garage door already close itself...

RavenSystem, 2019-02-22:
@pilot1981 I think timers are explain well in the wiki. Problem is that there is not a timer for auto-closing feature. In your particular case, you must use a Sonoff SV with close and open sensors. 

@GPL71 I will fix bugs asap. I hope this weekend.

pilot1981, 2019-02-22:
so, If I use a SONOFF SV, I have to:
- attach close sensor to GPIO14 (it's NC when garage door is closed)
- attach open sensor on GPIO4 (It's NC when garage door is opened)
- config Sensor Close to ON=NC
- config Sensor Open to 2=NC
- config Working Time 240 seconds is the time to open and close garage door itself

If I send open command in apple home app, after 240 seconds are spent, switch in apple home app return to closed state and only now I can send a new open command?

it's all right? I understood well?

GPL71, 2019-02-22:
@RavenSystem ... Thank you !!! you're a great. Please note that I discovered a second bug. I'd wrote
@Pilot1981 The SV is the best choice.... But if you have a door with autoself closing and logic board to manage safety of door (we have this systems in Italy e.g.) the basic with dry contact is enough for you....
This my opinion

pilot1981, 2019-02-22:
I agree, but I don’t understood howto transform Apple home app switch closed-open to a button only-open....

DV

RavenSystem, 2019-02-22:
* Working Time is the time necessary to open the door completely from a fully close status. It does not make sense if an open-sensor is used.

pilot1981, 2019-02-22:
but i apple home app I can have a button with only open command, without close command?

pilot1981, 2019-02-24:
So, when close sensor confirm garage door closed, the switch in Apple home app turn again to initial closed state?

dvelardi, 2019-02-25:
@RavenSystem

Today using SONOFF BASIC R2 cutted tracks, I found best config for my project; you can find here:

But I found a strange behavior:

When close sensor send signal, Apple home app remain opening state (wrong) and EVE app change state to close (right state).

It’s a bug?

Using SONOFF BASIC, opened sensor signal came from GPIO2? The same of closed sensor?

I don’t think using SONOFF SV I can solve this issue...

DV

RavenSystem, 2019-02-25:
Open sensor comes always from GPIO4, and close sensor from GPIO14/GPIO2 (Depending of board type): 
https://github.com/RavenSystem/esp-homekit-devices/wiki/RavenCore#8-garage-door

pilot1981, 2019-02-25:
@GPL71 you modified SONOFF SV removing resistor? You can share your electric schema? It’s very useful for me!

DV

pilot1981, 2019-02-25:
@RavenSystem
Thanks for reply, but you don’t answer on inconsistent state between Apple home app and EVE app: you have an answer for me?

DV

dvelardi, 2019-02-26:
@RavenSystem

I have this issue: when I send open command, Apple home app show garage door always in opening state and if I ask Siri which it’s the state of garage door, Siri say don’t know...

DV
3389305149<tel:3389305149>

link - PRIVACY POLICY. http://www.meadinformatica.it/privacy/

pilot1981, 2019-02-28:
@RavenSystem close sensor seem not work for me on SONOFF basic; if I ask garage door status, Siri always respond always closed...I attached magnetic contact to GPIO2 and GND...it's a bug?

dvelardi, 2019-02-28:
@RavenSystem I understood the problem: I’m using SONOFF BASIC R2: I suppose GPIO2 don’t work with your firmware because it’s different from GPIO14 of SONOFF BASIC R1

You agree?

You have a solution for me?

I can use RX pin to connect close sensor?

RavenSystem, 2019-02-28:
Please, read the Wiki.

closed 2019-02-28 by RavenSystem

pilot1981, 2019-02-28:
sorry, I didn't understand...where is my error? please explain...

RavenSystem, 2019-02-28:
From the Wiki:

[image]

pilot1981, 2019-02-28:
you're right! thanks 

sonoff new board = sonoff basic R2

I go soldering right pin ...and try again

pilot1981, 2019-02-28:
Ok, magnetic sensor now is attached to GND and GPIO3/RX, but EVE and Apple home app always say garage door is closed...what else it’s wrong?

pilot1981, 2019-03-01:
@RavenSystem New issue in 0.6.2: first time you push button relay start, the second time relay don’t work...only led flash

pilot1981, 2019-03-01:
@RavenSystem In 0.6.2 I notice a new strange behavior: when I push button first time relay start, when I push button second time relay don’t start and led flash...it’s expected behavior or a new bug?

pilot1981, 2019-03-01:
@RavenSystem for sonoff basic R2 (new layout) I have to use always board type 1? I also tried 2, 3 and 4 but this don't solve my issue on close contact...

pilot1981, 2019-03-01:
@RavenSystem sorry. I answer myself: I have to use board type 4!!!!

pilot1981, 2019-03-01:
@RavenSystem happy news! All work fine!!! Thanks a lot for your help!

Your firmware is fantastic: great job!