RavenSystem/esp-homekit-devices #255: Not working Basic and CH4 Pro

netRunner0, 2019-02-21:
My step:

Erase memory - ok
Reboot - ok
Flash - ok (4ch pro - 8285, Basic 8266)
Reboot - ok
Power on - ok
Connect WiFi - not connect (no ip adress)
Reboot - ok
Connect WiFi - ok
Save settings - ok

After that, the devices start to connect endlessly to wi-fi and do not work.

With the same steps I successfully flashed Basic 1 and TH-16

EEsssBBeee, 2019-02-21:
I am facing the same issue using Shelly 1, it flashes fine and then seems to have problems beginning from the join wifi page. I have noticed that long SSID cannot be joined - i cannot join "Strawberry Fields Forever" but can join (sometime) on to SSID "IOT" same network, same encryption only difference is the SSID. I have to turn the other SSID's off to get the device to connect to the IOT SSID, it appears to have a GUI issue and i cannot select the correct SSID and enter the password.

pilot1981, 2019-02-21:
Already tried to reboot again? 

DV

EEsssBBeee, 2019-02-21:
I have rebooted many times, I have been struggling with this for over a week :(

pilot1981, 2019-02-21:
Power supply 3,3v of programmer or external 110/220v?

Use external power supply!

DV

EEsssBBeee, 2019-02-21:
i am using 3.3v of programmer, I found the issue. I earlier hunch was right, If I switch of the widi with the long name it works, it seems that even if i am not trying to connect to that SSID it messes up the GUI and prevents it from doing its job. Thanks @pilot1981 for taking a look, I now have a bunch of homekit shelly's

netRunner0, 2019-02-22:
@EEsssBBeee  my netwirk name "One"
@pilot1981 i m flashing 4ch pro with external power supply (from 3.3 it is not flashed)

GPL71, 2019-02-22:
Hey boys ...
I had some similar issues flashing by windows.
Solved using mac.

You need download the firmware when device is connected to 3,3v.
230 cause damages to computer

EEsssBBeee, 2019-02-22:
I had no problem flashing the firmware, I tried using Mac, windows 10 and linux. the issue I saw was that of SSID with names that are too long to fit in the GUI window. The firmware appears not to work well at all with long wifi names.

netRunner0, 2019-02-22:
I just flashed a small switch 5-32v, at 8285, there is a similar situation. Already 3 devices turned into brick ...

netRunner0, 2019-02-22:
Using https://github.com/Batyushin/Sonoff-HomeKit

Basic - Works!

closed 2019-02-28 by RavenSystem