RavenSystem/esp-homekit-devices #256: request new feature to control roomba irobot without scheduler

pilot1981, 2019-02-22:
Hi,
I would ask to implement a new type of device type compatibile with roomba irobot to implement these features:

- schedule start to clean in the time (not all roomba robot have it)
- know status (charging, cleaning, stopped, ...)
- request to return to dock station
- control movements (in the future only for fun ;-)

you could connect ESP8266 to serial port of roomba irobot: I found many example on internet..

let me know

regards

RavenSystem, 2019-02-22:
From the wiki:

> This work is focused on specific devices, like several Sonoff and Shelly devices. Any suggestion is welcome, but only official Apple HomeKit characteristics will be considered (Those that work with stock Apple Home App), and custom HomeKit characteristics that use third-party Apps will be rejected.

pilot1981, 2019-02-22:
yes, I would control roomba using apple home kit....

i3laze, 2019-02-23:
HomeBridge has this plugin (or even two):
https://github.com/AlexanderBabel/homebridge-irobot-roomba

Unless you gonna were Sonoff/Shelly to iRobot, that’s the wrong thread.

pilot1981, 2019-02-23:
I don’t want use homebridge...

DV

AramVartanyan, 2019-02-23:
It is using sample Switch services with Battery status and some local logic. The switch is named iRobot and that’s it.
Good idea! 👍🏻

peros550, 2019-02-23:
Is it based on a esp8266?

pilot1981, 2019-02-25:
@AramVartanyan can you explain better your idea?

AramVartanyan, 2019-02-25:
It is not my idea. My curiosity took me inside this code:
https://github.com/AlexanderBabel/homebridge-irobot-roomba/blob/master/index.js
and I just share what I found inside. I believe it is worth a try.
But maybe it will be hard to do it with RavenCore. You will need custom code for it.

closed 2019-02-28 by RavenSystem

pilot1981, 2019-03-03:
@AramVartanyan you think it’s possibile use part of code of @RavenSystem and ESP HOMEKIT library to control iRobot roomba?
For me it’s sufficient have scheduler, start and Dock feature...
I don’t want add an homebridge...
I think to use ESP01 and buck converter to install onboard irobot roomba...
@RavenSystem we can use your code?

RavenSystem, 2019-03-03:
> @RavenSystem we can use your code?

You can read license terms here: 
https://github.com/RavenSystem/esp-homekit-devices/blob/master/LICENSE