RavenSystem/esp-homekit-devices #258: Shelly1 stateless switch not working on 230AC

gijsstegehuis, 2019-02-24:
I followed @GPL71's [instructions](https://github.com/RavenSystem/esp-homekit-devices/issues/202#issuecomment-457902592).
All is working fine until i hook it up to 230AC, because then it stops responding.

So when i configured everything on 3.3v, it seems to be all working fine.
Then when i hook it up to 230AC, it stops responding in the home app.
I then connected it back to 3.3v to check the console output, but that doesn't show anything at all.
Looks like it crashed somehow. Is there any way i can recover the log files from it?

It's worth noting that i didn't connect any load to the OUT-A and OUT-B ports on the shelly.

GPL71, 2019-02-25:
post config in EVE app

gijsstegehuis, 2019-02-28:
@GPL71 see screenshots below:

[image]
[image]

RavenSystem, 2019-02-28:
Device type 3? Why?

gijsstegehuis, 2019-03-01:
Hi @RavenSystem. I used device-type 3 because @GPL71 [suggested it here](https://github.com/RavenSystem/esp-homekit-devices/issues/202#issuecomment-457902592).
It seems to work fine on 3.3V, but it stops working when i hook it up to 230V.

The [link you are referring to](https://github.com/RavenSystem/esp-homekit-devices/wiki/RavenCore#1-switch-1ch) creates a regular switch. That works just fine on the Shelly1 btw :)

What i am looking for is to create a stateless switch (so a button actually) like the Socket + Button you get from device type 3. Is there another way to achieve this?

RavenSystem, 2019-03-01:
I think @GPL71 was talking about board type, but he wrote device type instead. He has a bunch of Shelly workings and he know very well how configure them. Actually, only device type 1 and 2 are supported with Shelly. Ravencore will not support Shelly with device type 3, but I am developing a new firmware called HomeKit Accessory Architect (HAA) that will let you create a full custom HomeKit accessory.

closed 2019-03-01 by RavenSystem

gijsstegehuis, 2019-03-01:
@RavenSystem thx, this clarifies things.
The HAA part sounds great. Sounds like that would allow us to run shelly as a stateless switch?
If you need any help testing that, then just let me know.

@GPL71 can you confirm you indeed meant setting the board type?
Did you get the Shelly working as a stateless switch on 230VAC with the settings you proposed?

RavenSystem, 2019-03-01:
> Sounds like that would allow us to run shelly as a stateless switch?

Yes. It is an open system, so you will can setup anything that you want.