RavenSystem/esp-homekit-devices #259: ESP-01 connected with relay

lizzus, 2019-02-26:
Hi, I sucessfully flash a ESP-01S (egual to ESP-01 but with 1MB mem) with Ravencore firmware and add to Home app. I also configured it with EVE app. 
I tryed to use without any relay but connecting a button to GPIO0 and GND no response on app. I also tryed to connect to a relay pcb (like this pic https://ibb.co/F6KpB51) but not work. Any suggestion?

ymyshko, 2019-02-27:
Connect ESP01 to UART and start app like Putty.
Connect to port and you can see what happen.
Default device type 1. ESP01 have to work. Without relay.
If you don't see anything, you have bad UART device.

lizzus, 2019-02-27:
ESP01 work without relay, but if I connect to relay not work.

yukonx, 2019-02-28:
https://github.com/RavenSystem/esp-homekit-devices/wiki/RavenCore#board-type-default-1

GPIO0 and GPIO2 must be free or high in order to let ESP device to boo

lizzus, 2019-02-28:
So what can I do?

peros550, 2019-02-28:
Maybe a resistor between Vcc and GPIO0 or GPIO2 would make them see High voltage during boot up. If that's not messing up with your rest design.

lizzus, 2019-02-28:
I’ll try

yukonx, 2019-02-28:
https://ibb.co/52g3JQ6

I am replacing I/O

lizzus, 2019-02-28:
I wait your response.

ymyshko, 2019-02-28:
> So what can I do?

ESP01 have two gpio0 and gpio2 pins.
gpio0 connected to relay. But gpio0 and gpio2 must be high when esp boot. 
You can do az yukonx - replace gpio0 to gpio12 on ESP01
or replace on your relay board gpio0 to RX and recompile project.

RavenSystem, 2019-02-28:
Please, read "board type" and "device type" in the Wiki.

lizzus, 2019-02-28:
@RavenSystem I don’t understand what do you mean...

RavenSystem, 2019-02-28:
Device type:
https://github.com/RavenSystem/esp-homekit-devices/wiki/RavenCore#10-esp01-switch--button

Board type: 
https://github.com/RavenSystem/esp-homekit-devices/wiki/RavenCore#board-type-default-1

closed 2019-02-28 by RavenSystem