RavenSystem/esp-homekit-devices #260: Is it allowed to sell devices with this software on it?

danio0701, 2019-02-27:
Hi, I want to ask: is it allowed to sell devices with this software on it? I do not gonna do that but I'm curious because in Poland is few companies which are doing this.

GPL71, 2019-02-27:
Surely nooooo!!!
Ravencore is free.
I hope the sell the device to the right price and not charged for firmware

danio0701, 2019-02-27:
They are charge for firmware and are offering flash device for money using this firmware :( I can provide here offer in comment below to this companies

danio0701, 2019-02-27:
https://allegro.pl/oferta/sonoff-basic-kompatybilne-z-apple-homekit-dom-7795226507
AB Medical Partner Aureliusz Basista Przesław 58d/10 72-005 Przecław NIP: 852-176-83-40 REGON 812405790

https://allegro.pl/oferta/inteligentne-gniazdo-sonoff-s26-wifi-apple-homekit-7725170522
Michał Rabe
HouseIQ
Plac Kaszubski 8/311
81-350 Gdynia
NIP: 586-177-70-23
sklep@houseiq.pl

Here they even offer flash sonoff using your firmwere for 3$
https://allegro.pl/oferta/zmiana-firmware-supla-esp-afe-tasmota-homekit-7685632372

closed 2019-02-27 by danio0701