RavenSystem/esp-homekit-devices #262: Sonoff Touch and basic not connect to WiFi

danio0701, 2019-03-01:
Hi, I have issue with Sonoff Touch ver 1.0 and Sonoff basic.

Sonoff Touch is on RevenCore v0.6.2 (Same issue was on RevenCore v.0.6.2)
Sonoff Basic is on RevenCore v0.6.1

My issue is that every time when any of this device will lose connection with wifi (router reboot for eg.) they are not reconnecting back to wifi. In the Home app is info "Not responding". Only option to connect them back is to reset them and configure from the beginning.

My config for Sonoff Touch in Eve app is Default - only changed is "external toggle 1" option to 1.
Sonoff Basic have all default settings - only changed "Init State SW" to 1

Does anyone have any idea? 

Many thanks for suggestions because my router is restarting every day at 6 am (sometimes was freezing and this resolved my router issues :) )

RavenSystem, 2019-03-01:
I think that your router is not a good friend of multicast transmissions. What router have you?

danio0701, 2019-03-02:
My router is Tenda AC6 with firmware V15.03.05.16_multi. Now I noticed that when I will restart sonoff when router wifi is available then all is ok or if I will connect to the sonoff captive network and choose manually network.

Is it possible to program for sonoff feature that it will try to search for wifi again every 5min? for eg. 180 seconds is looking for wifi SSDI saved in memory, next as always starts AP mode and after 5 min in AP mode is trying to look for the network again.

peros550, 2019-03-03:
I believe this fork has solved the AP lock issue with a nice workaround. It makes esp device to attempt to reconnect to last known SSID every few minutes.

https://github.com/AramVartanyan/esp-wifi-config

danio0701, 2019-03-03:
probably this will good solution but I do not know how to implement it :/ upload ready bin file was easy. From second hand I prefer to have OTA option to update the firmware as my sonoff switch are in not good accessible places :/

danio0701, 2019-03-03:
> I believe this fork has solved the AP lock issue with a nice workaround. It makes esp device to attempt to reconnect to last known SSID every few minutes.
> 
> https://github.com/AramVartanyan/esp-wifi-config

probably this will good solution but I do not know how to implement it :/ upload ready bin file was easy. From second hand I prefer to have OTA option to update the firmware as my sonoff switch are in not good accessible places :/ For now I back to original sonoff firmware and homebridge on raspberry as this was working better for me. I will read more about other solutions as MQTT with tasmota or supla because I do not want to have all on China servers 👿

RavenSystem, 2019-03-10:
@danio0701 how much time need your router to boot?

danio0701, 2019-03-10:
> @danio0701 how much time need your router to boot?

1-2 min

RavenSystem, 2019-03-10:
It's is less than 3 minutes. Are you sure wifi is available before 3 minutes?

https://github.com/RavenSystem/esp-homekit-devices/wiki/RavenCore#reconnecting-to-wifi-network
https://github.com/RavenSystem/esp-homekit-devices/wiki/RavenCore#entering-wifi-ap-mode

danio0701, 2019-03-10:
I'm sure. I test it and it takes 50 seconds to wifi be available again.

closed 2019-04-16 by RavenSystem