RavenSystem/esp-homekit-devices #263: Sonoff basic as momentary relay for doorbell

overflywingo, 2019-03-02:
Hi 

It is possible to configure a Sonoff Basic for a Doorbell ? 
 I mean connecting GND & GPIO14/2 to the doorbell push button and a momentary power (1 second) for the bell ringing ? 
Perhaps possible via Garage door option ?

gijsstegehuis, 2019-03-05:
@overflywingo did you try [device type 3: Socket + Button](https://github.com/RavenSystem/esp-homekit-devices/wiki/RavenCore#3-socket--button) ?

RavenSystem, 2019-03-10:
https://appleinsider.com/articles/18/07/18/apple-purges-doorbell-category-from-homekit-accessories-page

closed 2019-03-10 by RavenSystem

AramVartanyan, 2019-03-10:
That is not exactly true.

gijsstegehuis, 2019-03-11:
@RavenSystem It's removed from the page, but not from the HomeKit framework.
The [Netatmo HomeKit compatible doorbell](https://www.netatmo.com/en-gb/security/doorbell?force_locale=en-gb) is coming soon.

RavenSystem, 2019-03-11:
Netatmo is going to release a video doorbell. It is not the same.

AramVartanyan, 2019-03-11:
The only limitation is that the doorbell characteristic shell be used as an addition to another service. A switch for example, or a button, or a security system, or whatever you like.
It cannot be a stand-alone service.

RavenSystem, 2019-03-11:
@AramVartanyan Exactly, I am working on a lock mechanism and I will try to incorporate a doorbell characteristic, but that is not a doorbell accessory. Only video doorbells accessories are allowed.

AramVartanyan, 2019-03-11:
Yes! You are absolutely right!

RavenSystem, 2019-04-13:
Supported in v0.8.2.