RavenSystem/esp-homekit-devices #265: ESP-01 GPIO0 and GPIO2 config.

lizzus, 2019-03-03:
@RavenSystem I'm using a ESP-01S connected to a relay (relay v.4) with latest firmware. In your guide you write "For ESP01 with a button connected to GPIO0 and a relay connected to GPIO2". Actually my config is:
- Board Type = 2
- Device Type = 10
- Reverse SW1 = Yes (because with "NO" is inverse)
But i think is GPIO0 connected to relay and GPIO2 connected to a button". 
I checked and this is correct config. What do you think?

RavenSystem, 2019-03-10:
From [Board Type](https://github.com/RavenSystem/esp-homekit-devices/wiki/RavenCore#board-type-default-1):

> 2. Sonoff new board layout: Uses GPIO2 instead GPIO14. Device type 10 has swapped relay and button GPIOs.

So, try Board Type 2 instead.

closed 2019-03-10 by RavenSystem

lizzus, 2019-03-10:
Did you read what I wrote??? I’m using device type 2!!!!!! 
Correct configuration is GPIO connected to relay and GPIO2 connected to button!!!!
You must correct your wiki

RavenSystem, 2019-03-10:
Wiki is right:

> With Board type 1 (Default):
> Button connected to GPIO0 and relay connected to GPIO2.

> With Board type 2:
> Swapped button and relay GPIOs: Button connected to GPIO2 and relay connected to GPIO0.

netRunner0, 2019-10-20:
Board type 1/2 and device type 10 for esp01 do not work for me
How can I set up a lock in HAA on gpio0 or gpio2?

RavenSystem, 2019-10-20:
What do you want to connect on GPIO0 and GPIO2?

netRunner0, 2019-10-20:
[image]

netRunner0, 2019-10-20:
GPIO0

lizzus, 2019-10-20:
model you posted idn't compatibile, because not use GPIO 0 and 2!

netRunner0, 2019-10-20:
> model you posted idn't compatibile, because not use GPIO 0 and 2!

What does she use? There are only 2 GPIOs that go to the relay board

netRunner0, 2019-10-20:
[image]

lizzus, 2019-10-20:
this model is full compatible with Ravencore:
[image]
[image]
Check version: v4! v1 is not compatible.

netRunner0, 2019-10-21:
For this relay, you must use:
serial.write

instead

gpio.write

RavenSystem, 2019-10-21:
Serial control is not supported.

netRunner0, 2019-10-21:
Relay in the trash?