RavenSystem/esp-homekit-devices #266: ISSUE : Basic+TH Sensor <= doesn’t work

claire078, 2019-03-04:
Hi,
I have an issue with latest firmware. 
I use a Basic with DS18B20. I made two rules: if temp >=26C switch on, <=25C switch off. 
Switch ON rule works good, switch OFF doesn’t. 
It could be a bug?

claire078, 2019-03-04:
It’s working good in Thermostat mode. 👍

claire078, 2019-03-05:
One more thing to Thermostat mode.
In EVE app it show 32.1C actual temp. through internet. I think at home it works good.
Home app says "Temp is 25C, cooling to 25C" Thermostat switched off. It should say just "Temp is 25C", does it?

What is the switching different? 0.1C?

@RavenSystem : could you confirm these Problems?
Thank You!

RavenSystem, 2019-03-10:
Home App rounds at 0.5ºC:
- 24.7ºC will be displayed as 24.5ºC in Home App.
- 24.8ºC will be displayed as 25ºC in Home App.

And Siri doesn't use decimals:
- 24.4ºC will be 24ºC for Siri.
- 24.5ºC will be 25ºC for Siri.

But temperatures used to work with thermostat are always the real temperature. EVE App shows the real temperature.

claire078, 2019-03-10:
OK. 
Original problem has confirmed?

RavenSystem, 2019-03-10:
Your original problem is not a real problem. Those rules are triggered when AppleTV contacts with temperature sensor accessory, some times is fast, and others, no. Because that, is better to use a thermostat accessory.

claire078, 2019-03-10:
But it is working for in wan way why doesn’t another way? It’s not logical.

RavenSystem, 2019-03-10:
Yes, it is not logical. Some time ago, I used an EVE Room with temp sensor, and an EVE socket, to create a custom thermostat with rules (Note that both accessories are HomeKit certified). It was a mess, sometimes it worked, sometimes not.

closed 2019-03-17 by RavenSystem

lizzus, 2019-03-28:
I used device type 6 with a sonoff basic. All worked perfectly. I changed value of TH period from 30 to 10 and rebooted device. After reboot sensor not working... I restored th period to 30 and not work. What do you suggest?