RavenSystem/esp-homekit-devices #267: magichome

skop1986, 2019-03-04:
Hello, could you build the firmware for Magichome?

rgnovak, 2019-03-04:
Think this will be done in the new firmware "HomeKit Accessory Architect" which is in development.

RavenSystem, 2019-03-10:
Only Apple HomeKit, sorry.

AramVartanyan, 2019-03-10:
For Magic Home device you can use this:

https://github.com/AramVartanyan/esp-homekit-demo/tree/master/examples/magic_home_strip

RavenSystem, 2019-03-13:
Sorry, I was wrong. Magic Home is not Google Home. Sometimes I read issues too fast.

Jaguaza, 2019-03-16:
We can have support for LED modules in the near future. Magic home???

-------------
Podemos tener soporte para módulos LED en un futuro próximo. Magic home???

Me gustaría participar más de lleno en tu proyecto, puedo probar betas, tengo gran experiencia, espero contactes conmigo.

jorgetechtalents, 2019-03-21:
+1 jaguaza

epoppis, 2019-03-29:
+1 vote

tohmc, 2019-03-31:
+1 vote

RavenSystem, 2019-04-01:
I have a PoC with RGB and RGBW, and I will incorporate it to RavenCore. Because MagicHome LED devices uses different pins, I need some info about that. What is the highest pin used in any RGBW channel?

Jaguaza, 2019-04-01:
Esta es la información que dispongo. Si te hace falta algún módulo magic home avísame por privado y vemos cómo lo hacemos.

Extraída de tasmota.

MagicHome with ESP8285

Yet another MagicHome RGB controller. I specifically had this one: https://www.aliexpress.com/item/DC5-24V-Wireless-WIFI-LED-RGB-Controller-RGBW-Controller-IR-RF-Remote-Control-IOS-Android-for/32827253255.html

Aside from it going 9-28v, there are no other external model numbers etc, however opening it up revealed a very different configuration inside, using an ESP8285 chip directly on the board.

opened with cover in view
(Right click and view the original image to see the full size image for all images below)

Identifying this board

Looking closely you can see it's an ESP8285, on the main board, it has the part number: ZJ-WFMN-B V1.1

board top 1  board top 2
Flashing

I found that on the back it has pads to solder to so you can flash it.

board back
The IO0 port is GPIO0 that needs to be pulled to ground when powered on for flashing (as per all ESP devices). I soldered cables to each of these. Note that some are very close to other pads, so be careful. If you want to avoid soldering, also pogo pins work well with the pads.

board back with cables
Configuration

Once flashed, I configured it as follows (using Tasmota v6.1.1):

ZJ-WFMN-B V1.1

Field	Value
Module type	18 Generic
D1 GPIO5	38 PWM2
D6 GPIO12	39 PWM3
D7 GPIO13	40 PWM4
D5 GPIO14	37 PWM1
ZJ-WFMN-A V1.1

Field	Value	Function
Module type	18 Generic	Module type
D2 GPIO4	51 IRRecv	IR Remote (optional, view console for debugging!)
D1 GPIO5	38 PWM2	BLUE
D6 GPIO12	37 PWM1	RED
D7 GPIO13	39 PWM3	GREEN
D8 GPIO15	40 PWM4	WHITE
ZJ-WFMN-A V1.1(RGB version)

Field	Value	Function
Module type	18 Generic	Module type
D1 GPIO5	38 PWM1	RED
D6 GPIO12	37 PWM2	GREEN
D7 GPIO13	39 PWM3	BLUE
With all that done, one of the critical things I needed to do so I could set the colors was to run the command to set SetOption15 to 1. For me, this was sending the MQTT command (you will need to adjust for your config):

mosquitto_pub -t 'cmnd/rgbled/SetOption15' -m '1'
Once the above was done and the device rebooted, I could set the colors using commands such as (to turn on red and white full):

mosquitto_pub -h openhab.sf -t 'cmnd/rgbled/color' -m '#ff0000ff'
RF control

If you have the RF variant, you will want to configure the GPIOs like this:

  { "MagicHome RF",    // Magic Home RF (ESP8266) - (Arilux LC10)
     GPIO_USER,        // GPIO00 Optional Button
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     GPIO_LED1_INV,    // GPIO02 Blue onboard LED (optional)
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor0
     GPIO_ARIRFRCV,    // GPIO04 RF receiver input 
     GPIO_PWM2,        // GPIO05 RGB LED Green
     0, 0, 0, 0, 0, 0, // Flash connection
     GPIO_PWM3,        // GPIO12 RGB LED Blue
     GPIO_PWM4,        // GPIO13 RGBW LED White
     GPIO_PWM1,        // GPIO14 RGB LED Red
     GPIO_LED2_INV,    // GPIO15 RF receiver control
     0, 0
  },
After chosing the correct module type in your Tasmota configuration, press a key on the remote after boot for the device to learn your code.

RavenSystem, 2019-04-08:
@Jaguaza Please, can you test this beta version with RGB/RGBW support?
Use EVE to configure RGBW pins. If you don't want W pin, set it to 0.

[ravencore_0.8.0beta1.zip](https://github.com/RavenSystem/esp-homekit-devices/files/3054665/ravencore_0.8.0beta1.zip)

Jaguaza, 2019-04-08:
Ok esta noche lo pruebo en casa y le indico el resultado.

Jaguaza, 2019-04-09:
Hola buenos días.

Ayer tuve dos módulos uno RGB y otro RGBW.

Los dos funcionando perfectamente, el pin del w tuve que configurarlo en el 15.

La única pega que le encuentro es que las tiras pierden intensidad, se ve con menos lúmenes o brillo y que el dimmer lo realiza a escalones no es un dimmer suave.

Por lo demás muy buen trabajo. En el módulo RGB tuve que intercambiar los pines R por G, por el tipo de tira que tengo y todo esto sin problema, respuesta muy buena en encendido y apagado. Trabaja bien con las automatizaciones. Desde que llegue mi chica al medio día la tendrá encendida toda la tarde para comprobar si hace cosas raras.

RavenSystem, 2019-04-09:
Supported in RavenCore v0.8.0.

closed 2019-04-09 by RavenSystem

mafyata, 2019-04-15:
Hi , I am using v0.8.4 with magic home led strip controller. I use device type 14 for rgb/w. My controller is with ir 24 key remote, but after flashing the remote is not working. Other problem is that when I change brightness % sometimes it changes the color of the strip.

Jaguaza, 2019-04-15:
The remote control is not supported in ravencore.  You have to adjust the color boots in eve.  Cofigure the Color boots to 0% and check its operation after a restart.  
If it does not work raise the color boots until you get the best fit.

mafyata, 2019-04-15:
Will it be supported in future ? Can I download the original software for magic home ?

Jaguaza, 2019-04-15:
I can not answer that.  That's what the ravencore creator will have to say.  

Did you not make a copy of your firmware before flashing?

mafyata, 2019-04-15:
I did not make a copy.  Software version was 33.v6

Jaguaza, 2019-04-15:
I'm sorry but I do not have a device with command to go.