RavenSystem/esp-homekit-devices #268: Sonoff T1 R2 EU - only one button works

quartist, 2019-03-04:
Hello,

In Sonoff T1 R2 EU (ver. 1.0) - only one button work - button 1. HomeKit (and Eve app) see this model like SWITCH CH1. Button 2 does not work via app and without too. (When I touch button number 2 - not working). I try to debug this via screen and... when I touch button 1 all is ok - in logs I see changes 0 to 1 and 1 to 0, but when I touch button number 2 - log is empty - no reaction.
 Any ideas? This is my first 2CH Light switch.

quartist, 2019-03-05:
SOLVED!

in Eve APP I need to change device type to 2 and External Toggle 1 and 2 to 1 value.

Raven, could you add to wiki description about EVE parameter.

seritos, 2019-03-05:
Thanx @quartist for your suggestion. Got the same issue and now works perfectly

RavenSystem, 2019-03-10:
Clarified in Wiki. ;)

closed 2019-03-10 by RavenSystem