RavenSystem/esp-homekit-devices #272: Lets support @Ravencore with HW for his new developments and testings!

seritos, 2019-03-06:
Hi All,

For all of those who know about Ravencore, will understand that thanks to this incredible software, we are able to use accessories with homekit for almost a ridiculous amount of money. 
Many of us have introduced into the world of home automation due to the ease of use and have have been satisfied with his performance.

I know firsthand, that he is just about release a version that will suppot the window covering and door, this is, the long-awaited version  for the roller shutters!!

Is because of this new release in combination with the future FW (HAA), that I am reminding some support for him with some HW that he published on the Ravencore WIKI section (at the bottom of the page), that will directly take you to the amazon webite to be sent to him.

http://amzn.eu/02wuKCg

Thanx!

GPL71, 2019-03-06:
I agree with you 
@RavenSystem is doing the best and great free work available at moment
And I think that buy Hw form its link is a little efforts for us.

Than boys lets go using the link above becausa @RavenSystem had realized our dreams about domotica

apacherg, 2019-03-07:
Raven you're the number one ... Thanks

seritos, 2019-03-10:
Window rolling Shutter supported version published!!

GPL71, 2019-03-11:
@RavenSystem
Another great work with the support to device 12!!!
Thank you!!!

Please add Amazon links from italian store and I'll forward to our italian facebook page.

St0nehead, 2019-03-11:
Would be great if @RavenSystem adds Sonoff SC support..

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

seritos, 2019-03-11:
@St0nehead
This is a thread to help ravencore and supporting him as a reminder :) 

Thanx for understanding

RavenSystem, 2019-03-12:
@St0nehead Can you provide an email address, please?

St0nehead, 2019-03-12:
St0nehead999(at)gmail.com

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

RavenSystem, 2019-03-13:
@St0nehead did you receive my email?

St0nehead, 2019-03-14:
Yes! 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

awasspl, 2019-03-17:
@RavenSystem - Can I have the English version of "Your amazon wish list"? I can not understand how to buy anything:)

seritos, 2019-03-17:
@awasspl

You can also donate trough PayPal.

https://paypal.me/ravensystem

awasspl, 2019-03-17:
Yes, but i always think that paypal charges the fee.

seritos, 2019-03-17:
I dont think it does. Have a try!

GPL71, 2019-03-18:
Added the PayPal link in the page of our Italian Facebook Group
If the user choose 'send money to' then no fees are added to RavenSystem.

awasspl, 2019-03-18:
Just send it, hope all is on You're account :)

topzoe, 2019-03-21:
Hello José,
I sent you a donation on PayPal 😉 for your great  job.
Lilian PAYET.

GPL71, 2019-04-23:
@RavenSystems Josè 
Are arriving donation from Italian Group?
We remind it to all users

Did you receive emails from me?
We would like share something with all user of community

closed 2019-05-26 by RavenSystem