RavenSystem/esp-homekit-devices #273: ISSUES about GARAGE DOOR device 8

GPL71, 2019-03-06:
Hi @RavenSystem 
Thank you for update about garage door.
I don't use sensors
I'm having follow problems

1 - If I tap garage door, opening starts. a flywheel stars and neverending.
2 -If I say "Hey siri open garage door" it work, but if I say "Hey siri close garage door" the relay doesn't energize

I checked my setup with your (you'd post in a past issue) and it's the same.

Can you suggest me test or what's a possible my mistake?

Thank you
Gianluca

RavenSystem, 2019-03-07:
Wait a moment. What means “I don’t use sensors”? None? Close sensor is mandatory.

GPL71, 2019-03-07:
@RavenSystem .... Sensor is none in EVE. NOT used

RavenSystem, 2019-03-07:
Open sensor is optional, but close sensor is mandatory.

GPL71, 2019-03-07:
@RavenSystem
Thank you Jose .... Now it's clear.
Then .... better is if both sensor are installed... Mandatory is for closed sensor

i3laze, 2019-03-10:
As a temporary solution there’s a HomeBridge plugin 
https://github.com/iomax/homebridge-mqttgaragedoor that’s able to rely on timers only and work without sensors. 

However I recommend using either 3rd party reed sensor from Ali, or adding extra wires from existing garage door sensor to Sonoff GPIO14+GND. Internal garage controller circuit should run <=5V on that sensor for this to work.

RavenSystem, 2019-03-10:
Sensor is necessary because if you don't use it, HomeKit doesn't know real door status when it is opened with a RF remote or wall push-button. And the idea is the possibility to get notifications when your door is opened, for security purposes.

Obstruction sensor was implemented some time ago. Actually, when it is triggered, HomeKit system is notified and relay is disabled to avoid possible damages.

closed 2019-03-10 by RavenSystem