RavenSystem/esp-homekit-devices #275: (GARAGE DOOR) Keeping the current push button control operational

overflywingo, 2019-03-08:
HI 
I just received my Sonoff SV to use with my garage door.
My garage door is for now controlled with a simple push button .. and despite the fact that I'll put a Sonoff SV, I would like to keep that button operational.  

Where do I have to connect the 2 wires of the push button the Sonoff ? It's a bit unclear to me.

RavenSystem, 2019-03-09:
You can keep your current push button. That is the idea. You must connect 2 wires of your button to isolated Sonoff relay, keeping your current push button connected to the garage door driver too.

overflywingo, 2019-03-09:
Thank you, I understand that is in parallel. Right ?

Can you tell me onto witch pin of the Sonoff SV I have to connect the 2 wires of my button ?   In +  ? In - ? Out + ? Out - ? Or elsewhere ? 
Sorry for my ignorance

overflywingo, 2019-03-09:
Hello

Can someone validate my shema  ?

RavenSystem, 2019-03-09:
In+ and Out+ , in isolated mode.

RavenSystem, 2019-03-10:
@overflywingo your layout is right.

topzoe, 2019-03-10:
Very good layout! 
Thank's a lot!

> Hello Can someone validate my layout ? [image]
> […](#)

GPL71, 2019-03-11:
@overflywingo
Can I use it too in my Facebook page?
Me and @lizzus have an italian Facebook page dedicated to Ravencore firmaware and I translated Jose's wiki adding a my picture of sonoff SV.
This one is better

pilot1981, 2019-03-11:
@GPL71 please post your facebook URL page?

overflywingo, 2019-03-11:
@gpl71. Of course you can use it onto your FB page 😀

lizzus, 2019-03-11:
This is not a page, but a group. We talk about HomeKit devices and firmware. Link is: https://www.facebook.com/groups/homekititalia

closed 2019-03-11 by RavenSystem

jat80, 2019-06-18:
@overflywingo I'm having issues with my Sonoff SV and garage door setup issue is here:

https://github.com/RavenSystem/esp-homekit-devices/issues/428

How is your SV going?  I only have GPIO14 close sensor and not the open one but as you can read from above issue, it's not working properly.