RavenSystem/esp-homekit-devices #277: Question : Switching differential?

claire078, 2019-03-10:
Hi,
what are the switching differential/hysteresis by thermo sensors? DS18B20, DHT22
It could be adjustable with Ravencore?

RavenSystem, 2019-03-10:
What do you mean?

claire078, 2019-03-10:
I mean, if setpoint 25C where switches on? 24,5C or 24,9C? So hysteresis is 0.1C, 0.2C, 0.5C ? 
Would be great if we can adjust it?

RavenSystem, 2019-03-10:
I understand you are talking about Thermostat device. If you set thermostat to 25ºC, relay will activated when sensor reaches 25ºC, and relay will deactivated when sensor down to 24.9ºC (In heater case). Sensor is read every 30 seconds, but you can configure polling period time between 3 and 60 secs.

claire078, 2019-03-10:
By borth sensor typ same?
Could we adjust it?

RavenSystem, 2019-03-10:
> By both sensor typ same?

Yes.

Could we adjust it?

> No.

claire078, 2019-03-10:
Me 0.1C hysteresis is good for floor heating but at another heating mode would be better to adjust possibilities to change to 0.2 or 0.5C. 
Thanks! 👍

closed 2019-03-10 by claire078