RavenSystem/esp-homekit-devices #278: Command Not found

Crox77, 2019-03-10:
Hi looking for some help please>.

after typing Crox-iMac:~ Crox-iMac$ esptool.py -p /dev/usbserial-A9RK0041 erase_flash

I get this error in Terminal 

-bash: esptool.py: command not found
Don't really know what I'm doing and looking for some help please?

Thanks

topzoe, 2019-03-10:
Hi, 
you must install esptool...
In Terminal you type: pip install esptool

Crox77, 2019-03-10:
Hi thanks for your reply, I have already installed esp using that command

RavenSystem, 2019-03-11:
> -bash: esptool.py: command not found

@Crox77 You system is telling you that you have not installed esptool.py

Crox77, 2019-03-11:
Thanks for your reply, I have ran the pip install esptool and it says installed however I still get this error message

RavenSystem, 2019-03-11:
Exec: `find /usr/ -name esptool.py`

This command will search esptool.py in you /usr/ directory

Crox77, 2019-03-11:
comes back with this
find: /usr//sbin/authserver: Permission denied

RavenSystem, 2019-03-11:
Exec: `find / -name esptool.py`

This command will search esptool.py in your / directory

Crox77, 2019-03-11:
sorry to be a pain but tried that, and come back with the permission denied again.

Crox-iMac:~ Crox-iMac$ find / -name esptool.py
find: /usr/sbin/authserver: Permission denied

RavenSystem, 2019-03-11:
@Crox77 If after exec `find / -name esptool.py`and search in your entire disk, it does not show esptool.py path, for sure it is not installed in your system.

Crox77, 2019-03-11:
ive ran the install command pip install esptool again:
Crox-iMac:~ Crox-iMac$ pip install esptool
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Requirement already satisfied: esptool in ./Library/Python/2.7/lib/python/site-packages (2.6)
Requirement already satisfied: pyaes in ./Library/Python/2.7/lib/python/site-packages (from esptool) (1.6.1)
Requirement already satisfied: pyserial>=3.0 in ./Library/Python/2.7/lib/python/site-packages (from esptool) (3.4)
Requirement already satisfied: ecdsa in ./Library/Python/2.7/lib/python/site-packages (from esptool) (0.13)

this is what it comes back with but still get the command error.

what else can I try? Thanks

Crox77, 2019-03-11:
Search has found it on my system:

ind: /Users/Crox-iMac/Library/HomeKit: Operation not permitted
/Users/Crox-iMac/Library/Python/2.7/bin/esptool.py
/Users/Crox-iMac/Library/Python/2.7/lib/python/site-packages/esptool.py

Not sure what to do next

RavenSystem, 2019-03-11:
`export PATH=$PATH:/Users/Crox-iMac/Library/Python/2.7/bin`

Crox77, 2019-03-11:
Crox-iMac:~ Crox-iMac$ export PATH=$PATH:/Users/Crox-iMac/Library/Python/2.7/bin
Crox-iMac:~ Crox-iMac$

Crox77, 2019-03-11:
ok so now im getting this
    instance.open()
  File "/Users/Crox-iMac/Library/Python/2.7/lib/python/site-packages/serial/serialposix.py", line 268, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port /dev/usbserial-A9RK0041: [Errno 2] No such file or directory: '/dev/usbserial-A9RK0041'
Crox-iMac:~ Crox-iMac$ 

not sure I have the USB port number correct, I have installed the fdti drivers though

Crox77, 2019-03-11:
I think that is my FDTI Serial number not the port number, how do I find the USB port number?

St0nehead, 2019-03-12:
Type in your Terminal
`ls /dev`
Then paste result here

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Crox77, 2019-03-12:
sorry for late reply I've only just got in from work, anyhow after typing ls /dev I get this:

Crox-iMac:~ Crox-iMac$ ls /dev
afsc_type5			ptyw6
auditpipe			ptyw7
auditsessions			ptyw8
autofs				ptyw9
autofs_control			ptywa
autofs_homedirmounter		ptywb
autofs_notrigger		ptywc
autofs_nowait			ptywd
bpf0				ptywe
bpf1				ptywf
bpf2				random
bpf3				rdisk0
bpf4				rdisk0s1
console				rdisk0s2
cu.Bluetooth-Incoming-Port	rdisk1
cu.usbserial-A9RK004I		rdisk1s1
disk0				rdisk1s2
disk0s1				rdisk1s3
disk0s2				rdisk1s4
disk1				rdisk2
disk1s1				rdisk2s1
disk1s2				rdisk2s2
disk1s3				rdisk3
disk1s4				rdisk3s1
disk2				sdt
disk2s1				stderr
disk2s2				stdin
disk3				stdout
disk3s1				systrace
dtrace				tty
dtracehelper			tty.Bluetooth-Incoming-Port
fbt				tty.usbserial-A9RK004I
fd				ttyp0
fsevents			ttyp1
io8log				ttyp2
io8logmt			ttyp3
io8logtemp			ttyp4
klog				ttyp5
lockstat			ttyp6
machtrace			ttyp7
null				ttyp8
oslog				ttyp9
oslog_stream			ttypa
pf				ttypb
pfm				ttypc
profile				ttypd
ptmx				ttype
ptyp0				ttypf
ptyp1				ttyq0
ptyp2				ttyq1
ptyp3				ttyq2
ptyp4				ttyq3
ptyp5				ttyq4
ptyp6				ttyq5
ptyp7				ttyq6
ptyp8				ttyq7
ptyp9				ttyq8
ptypa				ttyq9
ptypb				ttyqa
ptypc				ttyqb
ptypd				ttyqc
ptype				ttyqd
ptypf				ttyqe
ptyq0				ttyqf
ptyq1				ttyr0
ptyq2				ttyr1
ptyq3				ttyr2
ptyq4				ttyr3
ptyq5				ttyr4
ptyq6				ttyr5
ptyq7				ttyr6
ptyq8				ttyr7
ptyq9				ttyr8
ptyqa				ttyr9
ptyqb				ttyra
ptyqc				ttyrb
ptyqd				ttyrc
ptyqe				ttyrd
ptyqf				ttyre
ptyr0				ttyrf
ptyr1				ttys0
ptyr2				ttys000
ptyr3				ttys001
ptyr4				ttys1
ptyr5				ttys2
ptyr6				ttys3
ptyr7				ttys4
ptyr8				ttys5
ptyr9				ttys6
ptyra				ttys7
ptyrb				ttys8
ptyrc				ttys9
ptyrd				ttysa
ptyre				ttysb
ptyrf				ttysc
ptys0				ttysd
ptys1				ttyse
ptys2				ttysf
ptys3				ttyt0
ptys4				ttyt1
ptys5				ttyt2
ptys6				ttyt3
ptys7				ttyt4
ptys8				ttyt5
ptys9				ttyt6
ptysa				ttyt7
ptysb				ttyt8
ptysc				ttyt9
ptysd				ttyta
ptyse				ttytb
ptysf				ttytc
ptyt0				ttytd
ptyt1				ttyte
ptyt2				ttytf
ptyt3				ttyu0
ptyt4				ttyu1
ptyt5				ttyu2
ptyt6				ttyu3
ptyt7				ttyu4
ptyt8				ttyu5
ptyt9				ttyu6
ptyta				ttyu7
ptytb				ttyu8
ptytc				ttyu9
ptytd				ttyua
ptyte				ttyub
ptytf				ttyuc
ptyu0				ttyud
ptyu1				ttyue
ptyu2				ttyuf
ptyu3				ttyv0
ptyu4				ttyv1
ptyu5				ttyv2
ptyu6				ttyv3
ptyu7				ttyv4
ptyu8				ttyv5
ptyu9				ttyv6
ptyua				ttyv7
ptyub				ttyv8
ptyuc				ttyv9
ptyud				ttyva
ptyue				ttyvb
ptyuf				ttyvc
ptyv0				ttyvd
ptyv1				ttyve
ptyv2				ttyvf
ptyv3				ttyw0
ptyv4				ttyw1
ptyv5				ttyw2
ptyv6				ttyw3
ptyv7				ttyw4
ptyv8				ttyw5
ptyv9				ttyw6
ptyva				ttyw7
ptyvb				ttyw8
ptyvc				ttyw9
ptyvd				ttywa
ptyve				ttywb
ptyvf				ttywc
ptyw0				ttywd
ptyw1				ttywe
ptyw2				ttywf
ptyw3				urandom
ptyw4				xcpm
ptyw5				zero
Crox-iMac:~ Crox-iMac$

RavenSystem, 2019-03-12:
`/dev/tty.usbserial-A9RK004I`

Crox77, 2019-03-12:
Thanks, Getting there managed to erase flash now get this when trying to re-flash:
Crox-iMac:~ Crox-iMac$ esptool.py -p /dev/tty.usbserial-A9RK004I --baud 115200 write_flash -fs 1MB -fm dout -ff 40m \
> 0x0 rboot.bin 0x1000 blank_config.bin 0x2000 otaboot.bin
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,40m,26m,20m,80m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'rboot.bin'

where do I need to put the 3 .bin files ive downloaded?

Crox77, 2019-03-12:
getting this now:
Crox-iMac:site-packages Crox-iMac$ esptool.py -p /dev/tty.usbserial-A9RK004I --baud 115200 write_flash -fs 1MB -fm dout -ff 40m \
> 0x0 rboot.bin 0x1000 blank_config.bin 0x2000 otaboot.bin
esptool.py v2.6
Serial port /dev/tty.usbserial-A9RK004I
Traceback (most recent call last):
  File "/Users/Crox-iMac/Library/Python/2.7/bin/esptool.py", line 2959, in <module>
    _main()
  File "/Users/Crox-iMac/Library/Python/2.7/bin/esptool.py", line 2952, in _main
    main()
  File "/Users/Crox-iMac/Library/Python/2.7/bin/esptool.py", line 2646, in main
    esp = ESPLoader.detect_chip(each_port, initial_baud, args.before, args.trace)
  File "/Users/Crox-iMac/Library/Python/2.7/bin/esptool.py", line 258, in detect_chip
    detect_port = ESPLoader(port, baud, trace_enabled=trace_enabled)
  File "/Users/Crox-iMac/Library/Python/2.7/bin/esptool.py", line 222, in __init__
    self._port = serial.serial_for_url(port)
  File "/Users/Crox-iMac/Library/Python/2.7/lib/python/site-packages/serial/__init__.py", line 88, in serial_for_url
    instance.open()
  File "/Users/Crox-iMac/Library/Python/2.7/lib/python/site-packages/serial/serialposix.py", line 268, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port /dev/tty.usbserial-A9RK004I: [Errno 2] No such file or directory: '/dev/tty.usbserial-A9RK004I'

Crox77, 2019-03-12:
Don't know what's going on now, my FTDI is no longer showing on my Mac, that's what the above message is, was all ok until I did the first erase flash

St0nehead, 2019-03-12:
Try to run command with ’sudo’

sudo esptool.py -p /dev/tty.usbserial-A9RK004l --baud 115200 write_flash -fs 1MB -fm dout -ff 40m 0x0 rboot.bin 0x1000 blank_config.bin 0x2000 otaboot.bin

closed 2019-03-24 by RavenSystem