RavenSystem/esp-homekit-devices #280: Device 12: Is shelly 2 or 2.5 compatibilty

GPL71, 2019-03-11:
@RavenSystem
Josè about your new great work .... Are shelly 2 or 2.5 compatible?
If no, when will they be?
Shelly is more little then a sonoff and more stable about wifi

Thank you

RavenSystem, 2019-03-11:
I want to support Shelly 2.5, including power measure to avoid motor damage. But I haven't one yet.

curla92, 2019-03-11:
we are waiting shelly functionality for rolling shutters

seritos, 2019-03-11:
@curla92
Ravencore has been working very hard to release this version. Currently has no shellys to test it. Meanwhile you can thank him for his work and support him with some hardware. 
The link is on the wiki section and in this forum some threads bellow.

curla92, 2019-03-11:
> @curla92
> Ravencore has been working very hard to release this version. Currently has no shellys to test it. Meanwhile you can thank him for his work and support him with some hardware.
> The link is on the wiki section and in this forum some threads bellow.

i know and I appreciate his work!

GPL71, 2019-03-12:
@RavenSystem
I' receive Shelly2 around first day of April... Then no problem to help for test. 
I think that power measuring for motor protection is genial..
Great work really!!!!!

GPL71, 2019-03-17:
@RavenSystem Josè a member of our italian Facebook page tried device 12 on shelly 2 and it work fine  with mount wall switch.
He found a bug with Siri control. If you ask Siri to open, shelly close and viceversa.
But when covering is full open or full closed the icon dispay the right position

curla92, 2019-03-19:
maybe the shelly 2 has the button in reverse. 
`            switch (external_toggle1.value.int_value) {
                case 1:
                    adv_toggle_create(button1_gpio, pullup);
                    adv_toggle_register_callback_fn(button1_gpio, covering_button_down, 0);
                    
                    adv_toggle_create(button2_gpio, pullup);
                    adv_toggle_register_callback_fn(button2_gpio, covering_button_up, 0);
                    break;
                    
                case 2:
                    adv_toggle_create(button1_gpio, pullup);
                    adv_toggle_register_callback_fn(button1_gpio, covering_toggle_down, 0);
                    adv_toggle_register_callback_fn(button1_gpio, covering_toggle_stop, 1);
                    
                    adv_toggle_create(button2_gpio, pullup);
                    adv_toggle_register_callback_fn(button2_gpio, covering_toggle_up, 0);
                    adv_toggle_register_callback_fn(button2_gpio, covering_toggle_stop, 1);
                    break;
                    
                default:
                    break;
            }`

if i change up and down It might work?
thanks

curla92, 2019-03-24:
hi josè. i fixed shelly 2 issue homekit. you can add this string `    homekit_characteristic_notify(&covering_target_position, HOMEKIT_UINT8(covering_target_position.value.int_value));
    homekit_characteristic_notify(&covering_current_position, covering_current_position.value);
    homekit_characteristic_notify(&covering_position_state, HOMEKIT_UINT8(covering_position_state.value.int_value));`

in the functions covering_button_up and covering_button_down. 
Let me know if you can update it. 
Thanks

RavenSystem, 2019-03-24:
@curla92 Many thanks.

There is a bug produced by a missing characteristic notification when external controls are used. But a better way to fix is adding `homekit_characteristic_notify(&covering_target_position, covering_target_position.value);` `into covering_on_callback()` func.

BTW it is better to use `covering_position_state.value`instead `HOMEKIT_UINT8(covering_position_state.value.int_value)`.

RavenSystem, 2019-04-13:
Shelly 2.5 GPIO map:
https://github.com/arendst/Sonoff-Tasmota/issues/5592#issuecomment-482151991

RavenSystem, 2019-04-13:
Shelly 2.5 will be supported soon, in RavenCore v0.8.3.

RavenSystem, 2019-04-14:
Shelly 2.5 supported in v0.8.3. Board type 5.

closed 2019-04-14 by RavenSystem