RavenSystem/esp-homekit-devices #284: RAVENCORE Y MODULOS LED.

Jaguaza, 2019-03-13:
Buenos días Jose.

Seguramente te habrán hablado ya de esto pero hoy quiero exponer por que el añadir este soporte nos ayudaría a mucha gente y dar soporte a que podamos tener un dispositivo mas en homekit a un precio reducido y con gran facilidad.

Se que el trabajo duro de picar código seria para ti, por eso pido a todos los interesados en este proyecto o bien donando a través de tu paypal o comprando los dispositivos necesarios para tu trabajo y sería la mejor forma de demostrarme que hay mucha gente que apoya tu trabajo y desea esta implementación en su firmware.

Como tu sabes la mayoría de gente solo quiere obtener la información que desea, obtener la respuesta deseada y no participar en un proyecto. Te digo esto por que mejor que tu sabrás  lo difícil que es que la gente se moje y eso es lo que he intentado yo en nuestro grupo de telegram para ver la gente que estaría interesada en añadir soporte led.

[https://drive.google.com/file/d/0Bz9UgMKPEfPJa0dKVnRHNlpBdWxxVWNlU01BTU85WVFzVHI4/view?usp=sharing](url)

Como ves en este enlace en menos de una semana han participado casi 90 personas y estarían interesadas un 65% de los votos en añadirlo.

Por que necesitamos de tu trabajo, bien un miembro del grupo realizo un binario para quemarlo en el esp del modulo que funciono en mi modulo y cuya información la extrajo de aquí, donde hay un ejemplo para led:

[https://github.com/maximkulkin/esp-homekit-demo](url)

Bien el problema es que los módulos no siguen un patrón de fabricación y que hay varios fabricantes y cada uno conecta el color a un gpio diferente.

Por ello el bin solo serviría para un modulo ya que no habría forma de seleccionar los gpio después de compilar.

 Bien por ello necesitaríamos tu ayuda y tu gran trabajo.

Otro motivo seria que dichos módulos son baratos y el flahseo de estos es relativamente fácil y seria una gran diferencia en comparación con marcas oficiales si no miremos a ver.

Precio de unas Philips hue de 1m.
[https://www.amazon.es/Philips-Hue-Lightstrip-extensi%C3%B3n-Compatible/dp/B07J4R7YZC/ref=sr_1_1_sspa?__mk_es_ES=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=2OW007GLQJL5J&keywords=philips+hue+lightstrip&qid=1552473765&s=lighting&sprefix=philips+hue+ligh%2Clighting%2C165&sr=1-1-spons&psc=1](url)

Ahora el precio con tu ayuda.

Modulo led wifi RGB o RGBW:
[https://es.aliexpress.com/store/product/16Million-colors-Wifi-RGB-RGBW-led-controller-smartphone-control-music-and-timer-mode-magic-home-mini/1514139_32865919702.html?spm=a219c.search0104.3.304.1fd27995u7oila&ws_ab_test=searchweb0_0,searchweb201602_5_10065_10068_10547_319_10891_317_10548_10696_10084_453_454_10083_10618_10307_10820_10821_10301_10303_537_536_10902_10059_10884_10887_321_322_10103,searchweb201603_6,ppcSwitch_0&algo_expid=fdec39b4-2fc0-4b9e-8ae5-46d20e7fe441-45&algo_pvid=fdec39b4-2fc0-4b9e-8ae5-46d20e7fe441&transAbTest=ae803_5](url)

Aquí los precios de la tira led para 5m tanto rgb como rgbw, no impermeable e impermeable.
[https://es.aliexpress.com/store/product/12-V-LED-Light-Strip-PC-5050-RGB-RGBW-RGBWW-12V-5M-60LEDs-m-Waterproof-Led/2955089_32907062381.html?spm=a219c.search0104.3.95.6f3a13a4WLFAAN&ws_ab_test=searchweb0_0,searchweb201602_5_10065_10068_10547_319_10891_317_10548_10696_10084_453_454_10083_10618_10307_10820_10821_10301_10303_537_536_10902_10059_10884_10887_321_322_10103,searchweb201603_6,ppcSwitch_0&algo_expid=22a90f64-8780-4e42-b00c-dc537c936255-14&algo_pvid=22a90f64-8780-4e42-b00c-dc537c936255&transAbTest=ae803_5](url)

Creo que la diferencia de precio es abismal.

Por otro lado tenemos la facilidad para flahsear ya que la carcasa es fácil de abrir, y las conexiones vienen bien visibles.
[https://drive.google.com/file/d/0Bz9UgMKPEfPJNTJYV2pxa3FZcGxqamtRSEQtd2QzOER5MUI4/view](url)

Bueno espero que esta propuesta llegue a buen puerto y nos ayudes a muchos usuarios a tener un soporte para nuestros módulos led.

greensouth, 2019-03-13:
+1, estoy a punto de recibir unos Shelly 2.5 estoy loco por probar Ravencore por primera vez y motorizar mis persianas con Homekit.

También tengo unos controladores MagicHome que se beneficiarían mucho de la propuesta. 

Enhorabuena por el trabajo y sigue así!

kalety, 2019-03-13:
Yes! I support this motion!

getafe83, 2019-03-13:
I am very interested in this option! Thank you!

DavidOliMar, 2019-03-13:
Thank you very much for your work. I hope to enjoy it soon.

sinuxo, 2019-03-13:
Seria genial, el poder tener esto. Me ha facilitado mucho ravencore y que se pudiera poner en tiras led sera la leche.

arturodiazad1, 2019-03-13:
Esto es algo que me interesa muchísimo, y que espero lo pueda incorporar pronto.

jorgetechtalents, 2019-03-15:
I think is a good idea because with that we Will be able to flash and have it in HomeKit Every thing that have a ESP

Jaguaza, 2019-03-15:
@ravencore o Jose.

Esta petición está hecha por muchos usuarios de un grupo de telegram.

Nos hubiese gustado una contestación, un simple por ahora no chicos. Tengo mucho jaleo más adelante quizás o simplemente nunca llegará esa incorporación al firmware de ravencore.

Bueno veo que tu tiempo escasea.

Simplemente gracias por tu trabajo y espero que continúe. 

Aunque no sirva de nada yo aporte mi granito e hice donación al igual que algunos de los que firmaron esta petición y aunque no hayamos tenido respuesta ánimo a los participantes a ella a qué donen bien dinero o hardware.

Muchísimas gracias y espero continues con este excelente trabajo.

Y por qué no algún día tener soporte a módulos led como lo tiene Tasmota o espurna.

Jaguaza, 2019-03-15:
@RaVeNcoRe or Jose.

This request is made by many users of a telegram group.

We would have liked an answer, a simple one for now no guys. I have a lot of fuss later perhaps or simply that incorporation to the ravencore firmware will never come.

Well I see that your time is short.

Just thanks for your work and I hope it continues.

Although it does not help I contribute my granite and I made a donation as well as some of those who signed this petition and although we have not had any response, encourage the participants to donate good money or hardware.

Thank you very much and I hope you continue with this excellent work.

And why not one day have support for led modules as has Tasmota or espurna.

pilot1981, 2019-03-16:
Which telegram group?

Jaguaza, 2019-03-16:
Telegram Group in Spanish: 

https://t.me/Apple_Homekit

RavenSystem, 2019-03-16:
@Jaguaza No suelo responder en hilos duplicados. De hecho lo que suelo hacer es marcarlo como duplicado, indicar el original, y cerrarlo; aunque este lo había dejado abierto. Por cierto, he visto tu donación, que ha sido la única. Muchas gracias.

RavenSystem, 2019-03-16:
https://github.com/RavenSystem/esp-homekit-devices/issues/267

closed 2019-03-16 by RavenSystem