RavenSystem/esp-homekit-devices #285: lock mechanism on garage door

pilot1981, 2019-03-13:
Hi,
different questions:
- in ravencore firmware, obstruction and open sensor are already operational?
- it's possibile attach to SONOFF SV to manage electromechanical lock mechanism?

RavenSystem, 2019-03-13:
> in ravencore firmware, obstruction and open sensor are already operational?

Yes. 

> it's possibile attach to SONOFF SV to manage electromechanical lock mechanism?

Yes, but better wait until device type 13 is released.

RavenSystem, 2019-03-13:
Dev type 13 released just now: 
https://github.com/RavenSystem/esp-homekit-devices/wiki/RavenCore#13-lock-mechanism

closed 2019-03-13 by RavenSystem

pilot1981, 2019-03-13:
Thanks for update and great job!

So if I would use garage door with lock mechanism you suggest to use sonoff SV with device type 7 together ESP01 with device type 13? How I can manage interaction/order to act between both devices?