RavenSystem/esp-homekit-devices #286: Customizing Ravencore

pandarek, 2019-03-13:
Hello,
I am trying to customize the Raven code to my needs and add a TH sens sensor to the Sonoff Dual R2. No change after modifying the code and adding Homekit characteristics. The HomeKit Server does not start. The software is restored and starts rehearsing. 
What should I know to give additional features to the devices?

RavenSystem, 2019-03-13:
It is a complex work. I suggest you that download Oficial Apple document for non-comercial HomeKit protocol from https://developer.apple.com . You will need to create a free developer account. 

And maybe a better way is play with examples from https://github.com/maximkulkin/esp-homekit-demo

pandarek, 2019-03-15:
Thank you, I think I solved the problem the device works :)