RavenSystem/esp-homekit-devices #288: Problem with reconnection to wifi after short unavailability of wifi network.

Basicek1, 2019-03-14:
After the OTA update to RavenCore v0.7.2 firmware, after a short unavailability of the wifi network, Sonoff Basic will not reconnect to the wifi again and will remain offline. The problem also appeared with Sonoff T1 R2 EU. Restarting the switch will help.
Please help and advice?
Thank you.

GPL71, 2019-03-14:
Did you try unplug and plug 230 Vac after few minutes?

Basicek1, 2019-03-14:
Thank you for your response.
Yes, I tried it. When disconnected from 220V and reconnected to 220V, it will connect to Wifi.
But if I turn off the WiFi AP again and plug it in again, the problem is repeated and the device will not connect to Wifi again.

RavenSystem, 2019-03-14:
The device connects wifi again, for sure. But you get “no response” status in Home App because mDNS announce is lost, because your router has been resetted. mDNS announce is send when device boots, and then every 4500 seconds. To prevent that, it is necessary to incorporate a watchdog that resend mDNS announce when device loses wifi connection and reconnects again.

Basicek1, 2019-03-14:
Thank you very much for your reply. You're right, the device has been given a DHCP server IP address. But the Home App is still "no response" despite being online for 5 hours. If I understand your answer correctly, is that not OK?

peros550, 2019-03-14:
> s. To prevent that, it is necessary to incorporate a watchdog that resend mDNS announce when device loses wifi connection and reconnects again.

is this fix going to be implemented in a future release of Ravencore?

St0nehead, 2019-03-14:
Just flip your plug (or swap the phase and zero).
This method works for me..

Basicek1, 2019-03-15:
1. José, thank you very much for your great project. I appreciate it. I am your big fan. ;-)
2. I would like to incorporate this (mDNS) fix into the next firmware. (Is this apparently implemented for certified Homekit devices?)
3. If you could still add support for Sonoff RF Bridge, it would be great ;-)
Best regards.

RavenSystem, 2019-03-17:
Next version will get a watchdog for wifi into mDNS subsystem. Code are ready. Every 15 seconds, network will be checked; if network downs, next time network is available, a new mDNS announce will be send.

nathsea, 2019-03-23:
Hi been having this issue today on the 4ch glad to see it is in the pipeline to be worked on. Thanks
Nath

RavenSystem, 2019-03-24:
Version 0.7.3 has the watchdog for mDNS. It has been released just now.

RavenSystem, 2019-03-26:
Any news about that using RavenCore 0.7.3?

peros550, 2019-03-26:
Did some tests yesterday. Basically, I had 3-4 devices running ravencore 0.7.3. My test case was just to toggle wifi on/off. The devices used to connect pretty much instantly when wifi was back. I'm going to simulate a power outage today on both the router and esp8266 and let's you know! 

Many thanks!

peros550, 2019-03-27:
Did more tests today. When the devices connect back to WiFi, they become available to home app almost instantly. 

When you incorporate Maxim's latest update of esp-wifi-config we can do more testing. Right now, there is a bug. If the esp8266 goes into AP mode, then the device never becomes again available in Home app. (tested today on 0.7.3)

Also, when you update esp-wifi-config I believe you can minimize the time out WIFI_CONFIG_CONNECT_TIMEOUT=180000.  That high value was used as a workaround for routers that needed more time to activate their AP. I'll keep an eye on your commits and try testing again. 

Thanks!

peros550, 2019-03-29:
@RavenSystem 

FYI
when you incorporate new version of esp-wifi-config you will need to use new version of init as per discussion [here ](https://github.com/maximkulkin/esp-homekit-demo/issues/191)

closed 2019-05-26 by RavenSystem