RavenSystem/esp-homekit-devices #289: Device 13 Thermostat

oigroigm, 2019-03-15:


lfiocco, 2019-03-20:
I ordered the Sonoff TH16, I will me testing this code on that device as a thermostat. I need it to control a pool heater.

I am not a programmer, But looking at the code (main.c) line 324 should be the trigger for the heater, line 331 should the line for the cool. 

Looks like the code is using INT variables. (whole numbers) 

You set the temp at 20.0 degrees. Real temp is 19.0. this will trigger the relay to switch on. Temperature starts to rise...

19.1
19.2
19.3 .....
19.9
20.0 <- this value now = the set point. the unit switches off..

Not sure about this part. As the temperature start to go back down, 19.9, 19,8 etc... will it trigger at 19.9, 19.5 or 19.0? if it triggers are 19.9 that's a problem. 19.5, not so bad. 19.0 would be prefect. that means that the dead band is 1 degree.

There should also be a timer between states. going from on/off can't happen if the last state was less than 10 minutes example. that means that once the unit switches off, it must wait 10 minutes before it switches on again. this is called compressor lockout.

I have some code that I modified to work on raspberry pi using Homebridge with a DS18B20 sensor. remember I am measuring water temp, a DTH22 will not work in water. let me know if you want it..it was based on this code.

https://github.com/ankurp/homebridge-pi-thermostat#readme

I will do some testing and let you know what I find. if it does trigger at 19.9, we can add 0.5 degrees to the calculations on those 2 lines.

Partizan7676, 2019-03-28:
Oigroigm, did you try device 5 ? It solves all your tasks. In eve app you set up temp offset and init state for your device. 
Btw see you working with mqtt? Trying to separate thermometer and thermo relay? I've done it with that firmware using th16 as thermostat and s20 as relay for heater. Rule was created with eve app. S20 power up when thermostat state is heating

lfiocco, 2019-03-28:
Can we please get a developer to explain the logic behind the temperature set point and sensors reading. There does not seem to be a dead band, the slightest change below and over will trigger on/off

lfiocco, 2019-03-31:
There is no dead band. So that reason I would not use this. It will damage equipment..I changed line 1524. 1000 - > 10000. This causes the polling range up to 600 seconds. You can’t see it but it works. I am using it to control a pool heater. Water does not change temperature fast so might work by delaying the poll cycle. I am testing it using a heat lamp on the sensor about a foot away. You should see it, always on/off, on/off. Within minutes of each other.

If you thing about it, thermostats controll the most expensive equipment in your home, which in turn controlls energy bills. What more expensive?

Using a simple < and > doesn’t cut it. You should be using more advanced logic. Something can be done with a few lines of code. I am not a programmer. Let’s hope!

RavenSystem, 2019-04-08:
Customizable deadband added in unreleased version 0.8.0.

lfiocco, 2019-04-08:
Nice thank you!!!

lfiocco, 2019-04-08:
You want me to test it?

RavenSystem, 2019-04-09:
Supported in RavenCore v0.8.0.

closed 2019-04-09 by RavenSystem