RavenSystem/esp-homekit-devices #290: Issue with "long press" in Mode 10 not working

overflywingo, 2019-03-15:
Hi
I've a Sonoff Basic v2 configured as " 10. ESP01 Switch + Button"

- The single press works well (in my set-up it activates another Sonoff)
- The double press works well (in my set-up it activates another Sonoff)
- ... but for the long press  it's not OK because it  didn't toogle the Sonoff socket physically, despite the fact that in the Home App I see well the related accessory icon changing state (on/off & off/on)
 
More info's : 
- board type 2
- device type 10
- firmware 0.6.2

Thanks for your help

RavenSystem, 2019-03-15:
Device type 10 works on a esp01 board, using gpio 0 and 2. It is incompatible with Sonoff devices.
https://github.com/RavenSystem/esp-homekit-devices/wiki/RavenCore#10-esp01-switch--button

overflywingo, 2019-03-15:
OK tanks,  so I have no solution to have a stateless button manage with an external toggle switch
I’ve tried with type 3, but unfortunately it seems that type 3 can’t manage external toggle switch

GPL71, 2019-03-15:
@RavenSystem Josè Is it possible have only a stateless button accessory?

RavenSystem, 2019-03-15:
Actually no. HAA will do it.

closed 2019-03-24 by RavenSystem