RavenSystem/esp-homekit-devices #291: Inching function and Lock system

GPL71, 2019-03-15:
@RavenSystem Josè thank you very much for device no 13 lock system!!! It work great!!!
I have some question and possible suggestions for feature.

1- Is it possible add a contact and notification if door is open or closed?
2- Is it possible add a custom timer? 
    Sometime you may have a lock mechanism a little bit "hard" and1 second relay closing is short
3- What do you think about follow situation
    In italy we have an inching button for lock mechanism and an inching button for stairs bulbs.
    Then .... Is possible have a firmware with lock and switch in the same board e.g. sonoff dual 7-32V?
    In my case for that pourpose I'm using a 7-32 v inching SONOFF and setted hardware inching on
    the board. It work perfectly, setting device 2. 
    Now ... is it possible custom only icons with one lock and one bulb?

Thank you again.....

RavenSystem, 2019-03-15:
1. That is planned. It is the next step.
2. Same that 1.
3. It will be possible with HAA, but RavenCore will not include that.

GPL71, 2019-03-15:
@RavenSystem Wonderful Josè .....
About 3 ... What does HAA mean?

RavenSystem, 2019-03-15:
New firmware HomeKit Accessory Architect.

GPL71, 2019-03-15:
I hope it will be soon .... I can't image what it will be

RavenSystem, 2019-03-24:
@GPL71 custom timer added in v0.7.3. You can choose from 0 to 5.1 secs (0 means 0.05 secs).

closed 2019-04-13 by RavenSystem