RavenSystem/esp-homekit-devices #293: Proble

blackwalkerhd, 2019-03-18:


blackwalkerhd, 2019-03-19:
Someone use the same sonoff model and flashed fine? 
If yes witch firmaware and now do you flash it?
Suggestions?

blackwalkerhd, 2019-03-23:
Sad no one is around :(

RavenSystem, 2019-03-23:
Logs, please.

blackwalkerhd, 2019-03-23:
Hi Raven, used ububtu to flash it. which log do you need? Tell me which one and i’ll try to get it as soon as possible, thanks a lot! :)

RavenSystem, 2019-03-23:
Erase and flash it again, try to setup it again, save entire log in a TXT file, and attach it here.

blackwalkerhd, 2019-03-23:
Ho well, i’ll do it tomorrow moorning! 
I live in italy and i like a lot your software! 
meanwhile I will see how and where to get the log file

RavenSystem, 2019-03-23:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Build-instructions#capture-logs

blackwalkerhd, 2019-03-24:
hi i write:
screen /dev/ttyUSB0 115200

blanck screen and nothing else....
trying to figure how capture the log...

blackwalkerhd, 2019-03-24:
[hardcopy.log](https://github.com/RavenSystem/esp-homekit-devices/files/3000788/hardcopy.log)
maybe i do it, can you check if i do it correctly?
thank a lot :)

RavenSystem, 2019-03-24:
That is your flash log. It's ok.

Now do it:
1.  Disconnect your device
2. Write `screen /dev/ttyUSB0 115200` but don't press enter key.
3. Connect your device and hit enter.

blackwalkerhd, 2019-03-24:
I capture  the log after writing the comand “screen” i press Ctrl+a and N

I have only to connect the device, type command and hit enter?

RavenSystem, 2019-03-24:
Connect the device, but don't flash it again.

blackwalkerhd, 2019-03-24:
done say screen is terminating

blackwalkerhd, 2019-03-24:
done again blank screen

RavenSystem, 2019-03-24:
I don't know where is the problem.

blackwalkerhd, 2019-03-24:
Disconnect your device
Write screen /dev/ttyUSB0 115200 but don't press enter key.
Connect your device and hit enter.

now i have blank screen with cursor up blinking

blackwalkerhd, 2019-03-24:
ok i close the system if you dont have idea, i can try again by myself and see how it go

blackwalkerhd, 2019-03-24:
thx anyway man :)

RavenSystem, 2019-03-24:
It seems that device is not sending data to UART.

RavenSystem, 2019-03-24:
Can you send a photo?

blackwalkerhd, 2019-03-24:
Sure, i’ll do later, its time to sleep for me, as i do it i’ll post it! Thx a lot ;)

closed 2019-04-13 by RavenSystem