RavenSystem/esp-homekit-devices #294: Is this switch supported?

400HPMustang, 2019-03-18:
Sorry for what may be a redundant question, but is the switch linked below supported by RavenSystem? and if so, do I need anything else to flash it or can it be done over USB or wifi?

https://www.itead.cc/inching-self-locking-wifi-wireless-switch.html

GPL71, 2019-03-18:
yes just tested .... inching function work too

pilot1981, 2019-03-18:
which device type we have to use for it? device type 10?

400HPMustang, 2019-03-18:
> yes just tested .... inching function work too

Does this require additional hardware to flash? or can I do it wirelessly or just hook it up to USB?

> which device type we have to use for it? device type 10?

I believe so. Using it as a wifi enabled button to garage door in a fashion similar to the wired garage door button.

GPL71, 2019-03-18:
@400HPMustang @pilot1981
- Use device 1
- you'll need to solder 2 wires on the esp for TX and RX. type 1 channel self/inching in internet an you'll find image
- you'll need a FTDI and ravencore  firmware

Remember pressing button on the board you'll have inching or switch function

closed 2019-04-13 by RavenSystem