RavenSystem/esp-homekit-devices #295: Need help to build custom firmware .bin

VannVann1992, 2019-03-18:
Hello,
Currently i'm trying to make a custom firmware using your Build instructions wiki
Because i just need to change the pairing pass code "021-82-017" to another one, and set the LOG OUTPUT to 0 by default
I stuck at the step 2, i just finish the esp-open-sdk install and i have the xtensa-lx106-elf/bin

And now the last step is "edit your PATH and add the generated toolchain bin directory. The path will be something like /path/to/esp-open-sdk/xtensa-lx106-elf/bin "
Can you please help me to change the PATH, which file should i edit in?

Thank you so much !

RavenSystem, 2019-03-18:
Changing pair code is illegal!! Read license.

VannVann1992, 2019-03-18:
I am so sorry because i don't know about that.
Please forgive me, i am sure i will not change it.
But i hope you can give me some more information about the path, because i really need it to change the LOG OUTPUT and make the default type to 2, because i have only Sonoff T1 2CH 
Thank you and sorry once again ...

closed 2019-03-20 by VannVann1992