RavenSystem/esp-homekit-devices #296: Acceesory category code

GPL71, 2019-03-19:
@RavenSystem Hi Josè ...
I'm looking for this repo of Maxim Kulkin 
https://github.com/maximkulkin/esp-homekit

I would like try generate the qr code using this tool to pair an accessory.
What's the number of CI when the device announce itself at the end of OTA setup?

Thank you

RavenSystem, 2019-03-19:
I am not sure, but I think that they are (from https://github.com/RavenSystem/esp-homekit-devices/blob/master/external_libs/homekit/include/homekit/types.h):

```
    homekit_accessory_category_other = 1,
    homekit_accessory_category_bridge = 2,
    homekit_accessory_category_fan = 3,
    homekit_accessory_category_garage = 4,
    homekit_accessory_category_lightbulb = 5,
    homekit_accessory_category_door_lock = 6,
    homekit_accessory_category_outlet = 7,
    homekit_accessory_category_switch = 8,
    homekit_accessory_category_thermostat = 9,
    homekit_accessory_category_sensor = 10,
    homekit_accessory_category_security_system = 11,
    homekit_accessory_category_door = 12,
    homekit_accessory_category_window = 13,
    homekit_accessory_category_window_covering = 14,
    homekit_accessory_category_programmable_switch = 15,
    homekit_accessory_category_range_extender = 16,
    homekit_accessory_category_ip_camera = 17,
    homekit_accessory_category_video_door_bell = 18,
    homekit_accessory_category_air_purifier = 19,
    homekit_accessory_category_heater = 20,
    homekit_accessory_category_air_conditioner = 21,
    homekit_accessory_category_humidifier = 22,
    homekit_accessory_category_dehumidifier = 23,
    homekit_accessory_category_speaker = 26,
    homekit_accessory_category_sprinkler = 28,
    homekit_accessory_category_faucet = 29,
    homekit_accessory_category_shower_head = 30,
```

GPL71, 2019-03-19:
@ Thank you Josè 
I'll try and if it's ok I'll share the .png labels

pandarek, 2019-03-22:
It seems to me that you need "setupId" to generate qrcode in homekit_server_config_t

GPL71, 2019-03-22:
@RavenSystem
because in the beginning when device is flashed it's a switch, I think th category is 8.
It seem so because log show the MDS announcement as ci=8.
Now I want try print a label using Maxim Kulkin routine and test on a device.
If it will work I'll share label with the community
Thank you

RavenSystem, 2019-03-24:
@GPL71 Yes, initially it is homekit_accessory_category_switch = 8.

GPL71, 2019-03-25:
@RavenSystem
Question .... The follow part of code is present in your firmware? 
Because only if that is present then QR system works

homekit_server_config_t config = {
    .accessories = accessories,
    .password = "123-45-678",
    .setupId="1QJ8",
};

Thank you

RavenSystem, 2019-03-25:
SetupId is missing, but I can add it.

GPL71, 2019-03-25:
@RavenSystem 
It would be great!!!!!
Thank you

GPL71, 2019-04-12:
@RavenSystem 
Josè is presente the setup id in firmware 0.8.1?

RavenSystem, 2019-04-12:
@GPL71 Ops... I forgot it, sorry. I am going to add it just now.

RavenSystem, 2019-04-13:
@GPL71 setupId added in v0.8.2.

GPL71, 2019-04-17:
@RavenSystem Great Josè

closed 2019-05-26 by RavenSystem

Marco-DE, 2019-09-09:
Hi @RavenSystem,

I flashed a Shelly1 on OTA so I could put it in the homekit. This works great!
How can I include multiple shellys in the Homekit? How can I change the Homekit ID or code? 
Can I change it somehow after flashing? If so, how exactly?

Marco

mafyata, 2019-09-09:
You do not need to change anything.

Marco-DE, 2019-09-09:
Hi,

 

Thank you that helped me a lot :-) And it's really great!

 

Also the following question. With the Eve app I can set the settings of the shelly. As described here: https://github.com/RavenSystem/esp-homekit-devices/wiki/RavenCore I tested it with 12V and everything works great. In the manual under the point Log output [Default: 1] it is recommended to set it to "0" at 240V. When I do that the SW switch on the shelly doesn't work anymore. And if I reset the settings of the log output [Default: 1] to 1 and reboot it is always set to 0. I had to flash the shelly ne to make it work again. What exactly is the function "Log output" intended for?

 

 

Thanks Marco

 

 

 

Gesendet: Montag, 09. September 2019 um 11:12 Uhr