RavenSystem/esp-homekit-devices #298: esp HomeKit pushes notification

topzoe, 2019-03-20:
Hello,
is it possible to make a new device like a motion sensor that pushes a notification triggered by a digital input?

Like there: http://www.kevindarrah.com/wiki/index.php?title=TrigBoard

Regards

RavenSystem, 2019-03-24:
Yes, of course. I want to add those kind of accessories, but I need more free time to code.

RavenSystem, 2019-04-12:
Planned for version 0.9.0.

topzoe, 2019-04-13:
Great ! 👍🏾😁

peros550, 2019-04-13:
Guys you have amazing ideas!

GPL71, 2019-04-23:
@RavenSystem Hi Josè
it possible build the device with follow choose and boards

BOARDS
NODE MCU
For more sensors in a single board 
You can set number and type of sensors by EVE setup
You can set or no an output for ralay

ESP 01
You can set only on sensor type by EVE setup
You can set or no an output for ralay

LIST OF SENSORS
Motion
Occupancy
Leak
Alarm
Anemometer
Rain sensor (only yes or no)

What do you think?

RavenSystem, 2019-05-26:
RC will not get this finally. HAA will do it.

closed 2019-05-26 by RavenSystem

pilot1981, 2019-06-19:
We wait for HAA feature...

Using RC, We can use sonoff basic with dummy switch and only sensor connected to GPIO?